esp8266/Arduino #5097: Issue/question about documentation

GTP95, 2018-08-31:
Here https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html is stated that "SPIFFS.open and dir.openFile functions return a File object. This object supports all the functions of Stream, so you can use readBytes, findUntil, parseInt, println, and all other Stream methods" but here https://www.arduino.cc/reference/en/language/functions/communication/stream/ I don't find any function to write data, not even the println that is mentioned in your documentation, but my Arduino IDE compiles correctly a code where I call write on a File object (but I couldn't test it yet). So either there are undocumented functions or I'm looking at the wrong place, could you please point me to a webpage where all Stream's functions are listed?
Thank you
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

CZEMacLeod, 2018-08-31:
@GTP95 If you read the description of File under the link you supplied you would see your answer:
> For functions like print(), Stream inherits from the Print class.

It is pretty simple to look up the source to see the members of Stream and Print - they are implemented by the ESP8266 Core in this repository.

[Stream.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Stream.h)
[Print.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Print.h)
[How to Use the Print Class](http://playground.arduino.cc/Code/Printclass)

I imagine this issue will get closed quickly as this is not really the place for questions like this - try stackoverflow or google - but if you find a mistake in the documentation or a bug in the implementation raise it here.

Hope this helps.

devyte, 2018-08-31:
Previous comment is correct in pretty much everything.
Closing.

closed 2018-08-31 by devyte