esp8266/Arduino #5100: Xtensa absolute filename too long

Elektrik1, 2018-09-01:
Hi there,

First compile on Windows 10 Arduino IDE 1.8.6 works fine, but on recompile such error appears:
Compiling core...

> cmd.exe /c mkdir "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\core" & (echo #define ARDUINO_ESP8266_GIT_VER 0x00000000 & echo #define ARDUINO_ESP8266_GIT_DESC win-2.5.0-dev ) > "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\core\\core_version.h"
> A subdirectory or file C:\Users\Elektrik\AppData\Local\Temp\arduino_build_391145\core already exists.
> 
> Using precompiled core: C:\Users\Elektrik\AppData\Local\Temp\arduino_cache_896884\core\core_arduino-esp8266_esp8266_nodemcuv2_CpuFrequency_80,VTable_flash,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_921600_4e056916d9045c6bc547a3a7ce5e20ae.a
> Linking everything together...
> "C:\\Dropbox\\Arduino\\Sketches\\hardware\\arduino-esp8266\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "C:\\Dropbox\\Arduino\\Sketches\\hardware\\arduino-esp8266\\esp8266/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145/eagle.app.v6.common.ld"
> "C:\\Dropbox\\Arduino\\Sketches\\hardware\\arduino-esp8266\\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -Wl,-Map "-Wl,C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145/HTTP-server.ino.map" -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-LC:\\Dropbox\\Arduino\\Sketches\\hardware\\arduino-esp8266\\esp8266/tools/sdk/lib" "-LC:\\Dropbox\\Arduino\\Sketches\\hardware\\arduino-esp8266\\esp8266/tools/sdk/ld" "-LC:\\Dropbox\\Arduino\\Sketches\\hardware\\arduino-esp8266\\esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" -Teagle.flash.4m1m.ld -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read -o "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145/HTTP-server.ino.elf" -Wl,--start-group "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\sketch\\HTTP-server.ino.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESP8266WiFi\\ESP8266WiFi.a" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\dscKeybusInterface\\dscKeybusInterface.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\dscKeybusInterface\\dscKeybusPrintData.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\dscKeybusInterface\\dscKeybusProcessData.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESP8266mDNS\\ESP8266mDNS.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ArduinoOTA\\ArduinoOTA.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncTCP\\AsyncPrinter.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncTCP\\ESPAsyncTCP.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncTCP\\ESPAsyncTCPbuffer.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncTCP\\SyncClient.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncTCP\\tcp_axtls.c.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncWebServer\\AsyncEventSource.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncWebServer\\AsyncWebSocket.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncWebServer\\SPIFFSEditor.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncWebServer\\WebAuthentication.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncWebServer\\WebHandlers.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncWebServer\\WebRequest.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncWebServer\\WebResponses.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\ESPAsyncWebServer\\WebServer.cpp.o" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145\\libraries\\Hash\\Hash.a" "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_cache_896884\\core\\core_arduino-esp8266_esp8266_nodemcuv2_CpuFrequency_80,VTable_flash,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_921600_4e056916d9045c6bc547a3a7ce5e20ae.a" -lhal -lphy -lpp -lnet80211 -llwip2 -lwpa -lcrypto -lmain -lwps -lbearssl -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group "-LC:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_391145"
> *xtensa-lx106-elf-gcc: error: C:\Users\Elektrik\AppData\Local\Temp\arduino_cache_896884\core\core_arduino-esp8266_esp8266_nodemcuv2_CpuFrequency_80,VTable_flash,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_921600_4e056916d9045c6bc547a3a7ce5e20ae.a: No such file or directory*

File is present but it can't be opened. I think it's because of long file name (>255 chars), since I cannot open this path with any text editor:
> C:\Users\Elektrik\AppData\Local\Temp\arduino_cache_896884\core\core_arduino-esp8266_esp8266_nodemcuv2_CpuFrequency_80,VTable_flash,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_921600_4e056916d9045c6bc547a3a7ce5e20ae.a

Could file name be shortened ?

Elektrik1, 2018-09-01:
File name is not associated with TEMP variable, and consists 208 chars in this case:
> core_arduino-esp8266_esp8266_nodemcuv2_CpuFrequency_80,VTable_flash,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_921600_4e056916d9045c6bc547a3a7ce5e20ae.a

Why can't it be shorter like "esp8266_4e056916d9045c6bc547a3a7ce5e20ae.a"  ?

d-a-v, 2018-09-02:
> File name is not associated with TEMP variable

it [seems to](https://superuser.com/questions/1079017/is-there-an-environment-variable-for-c-users-username-appdata-local-temp-in-w).

> Why can't it be shorter like "esp8266_4e056916d9045c6bc547a3a7ce5e20ae.a" ?

We don't control the arduino builder. After googling, it is advised to both change `%TEMP%` and give `arduino-1.9beta` a try. Can you try changing `%TEMP%` to something short ?

devyte, 2018-09-02:
Is the issue the filename length or the length of path+filename?

devyte, 2018-09-02:
On windows the max of path+filename is 260 (by default, the max can be disabled in the registry), so @d-a-v is right: change temp to point elsewhere.

5chufti, 2018-09-02:
> C:\Users\Elektrik\AppData\Local\Temp\

is the windows standard setting for %temp% and it is dangerous to recommend changing it as it may have ill sideeffects.

It is more likely to be cureable by using less "extra settings" with shorter names and values as long as Arduino IDE doesn't change the way of naming files (put extra settings in filename),

devyte, 2018-09-02:
In that case I suggest editing the registry to allow longer path+filename length, and test whether the Arduino IDE works with that.

5chufti, 2018-09-02:
I am not sure how many applications will be able to handle filenames longer than the traditional maximum. I wouldn't be surprised if windows itself has problems because this "novelty" is not respected in every aspect.

Elektrik1, 2018-09-02:
PATH edit is not a great solution, since the *filename* is the issue. 

Yes, windows can allocate 260 max chars absolute path, it's normally more than enough if reasonable filenames are selected. Filename itself is limited to 260 too, in my case it does not reach this limit yet, but if more options will be available it can exceed and this limit. 

Non elegant, more like hack setting `Enable win32 long paths` in gpeditor or via registry does not fix this issue, perhaps xtensa compiler itself cannot handle such long paths/filenames.

Temporarily manually deleting .a file between compilation helps, but full recompile is done every time.

d-a-v, 2018-09-02:
I have an unrelated coming update of the board generator. I'll try to reduce the internal option names in it since windows use them as part of the library name (which is odd?).

5chufti, 2018-09-03:
I don't think it is about uniqueness - the random part at the end should suffice, rather about detecting changed build options? Why this wouldn't be necessary under unix - I don't know.  Or maybe it is just an error in the Windows version?
maybe an occasional visitor (@PaulStoffregen) has more insight into the Arduino buildsystem?

d-a-v, 2018-09-03:
@5chufti does arduino-1.9 on windows has the same behavior ?

Elektrik1, 2018-09-03:
Yes, 1.9 beta has exact behavior.
Seems to be arduino issue indeed, turining off `Aggressively cache compiled core` in preferences fixes arduino_cache_xxx folder creation and caching of .a file.

d-a-v, 2018-09-03:
@Elektrik1 Can you try it ?

5chufti, 2018-09-03:
ok, disabling the _Aggressively cache compiled core_ removes the options from the filename and should aleviate Elektrik1's problem. 
This setting does not prevent the IDE from rebuilding everything when as little as the upload baudrate is changed [but this is an other (IDE) issue] so one can leave it disabled.

PaulStoffregen, 2018-09-03:
> maybe an occasional visitor (@PaulStoffregen) has more insight into the Arduino buildsystem?

Not really.  Truth is I've not followed the build system stuff closely since they moved it out of the Java codebase and made the Google Go language arduino-builder program.

But I do believe these issues on Arduino's repositories are related:

https://github.com/arduino/Arduino/issues/7902

https://github.com/arduino/arduino-builder/issues/284

I might also suggest a short-term workaround to at least ease user suffering.  Seems like many of the internal strings ESP8266 uses (in boards.txt) to define the menus are unnecessarily long.  Simply shortening those names could bring a lot of short-term relief while waiting for the Arduino developers to improve the builder.

d-a-v, 2018-09-03:
@PaulStoffregen Thanks for confirming that there is currently no other way than this short-term workaround.
(It is quite annoying since we hardly recognize anything in our boards.txt :)

d-a-v, 2018-09-03:
@Elektrik1 Please use latest git and report back whether the fix is enough for this issue

yoursunny, 2018-09-03:
I've been affected by this issue for two weeks. I have Arduino 1.8.6 installed from Windows Store and "Aggressively cache compiled core" is turned on. It seems that cc0bfa04d401810ed3f5d7d01be6e88b9011997f fixes this.

Elektrik1, 2018-09-03:
I do confirm latest commit fixes it, log:

Compiling core...

> cmd.exe /c mkdir "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_505278\\core" & (echo #define ARDUINO_ESP8266_GIT_VER 0x00000000 & echo #define ARDUINO_ESP8266_GIT_DESC win-2.5.0-dev ) > "C:\\Users\\Elektrik\\AppData\\Local\\Temp\\arduino_build_505278\\core\\core_version.h"
> A subdirectory or file C:\Users\Elektrik\AppData\Local\Temp\arduino_build_505278\core already exists.
> 
> Using precompiled core: C:\Users\Elektrik\AppData\Local\Temp\arduino_cache_703280\core\core_arduino-esp8266_esp8266_nodemcuv2_xtal_80,vt_flash,eesz_4M1M,ip_lm2,dbg_Disabled,lvl_None____,wipe_none,baud_921600_4e056916d9045c6bc547a3a7ce5e20ae.a
> Linking everything together...

Thank you!

d-a-v, 2018-09-03:
@Elektrik1 thanks for reporting !

closed 2018-09-03 by d-a-v

jjsuwa, 2018-09-20:
Now, **1.8,7** for Windows (or later) makes cached core `.a` filename MD5-hashed. Solved completely.