esp8266/Arduino #5101: Esp8266 wdt reset (UDP messaging)

nikrath, 2018-09-01:
Hello everyone.
I am using a nodemcu 1.0 (ESP-12E). All my code does is send some sensor data to any device that asks for it through UDP. For ex: I have an android app that sends a "GetStat" string to the board **every 1 second**, and upon detecting this string, the nodemcu sends a string in response. Here is the code:

```
void setup() {
  Serial.begin(115200);
  while(!Serial){;}//wait for Serial to begin
  Serial.println();
  
  if (firstTime() == false){ //if this is NOT the first time boot
    Serial.println("Found User SSID and Password in memory");
    WiFiInit();
  }
  Udp.begin(localPort);
  
  delay(2000); //wait 2 seconds
  Serial.println("Initialized");

  //INITIALIZING ALL THE PINS--->
  for(int i = 0 ; i<sizeof(pins)/sizeof(uint8_t) ; i++){
    pinMode(pins[i] , OUTPUT);
  } 
}

bool firstTime(){
  //FUNCTION TO CHECK IF THIS IS THE FIRST TIME OF BOOT
  //Basically I am trying to check if a username exists in the EEPROM
  char *cred = readChar(2 , strlen("photon photon"));
  if(strcmp(cred , "username password") == 0){
    //means this is indeed the first time of boot
    return true;
  }
  else{
    //means this is NOT the first boot
    int len = readInt(2); //read the length of the char that follows
    cred = readChar(4 , len);
    String s = split(cred)[0];
    String p = split(cred)[1];
    s.toCharArray(ssid , s.length()+1); //set the ssid to user's ssid
    p.toCharArray(pass , p.length()+1); //set the password to user's password
    return false;
  }
}

void loop() {
  packetSize = Udp.parsePacket(); //looks at network and sees if there is a request, and returns a size (zero if not request)
  if (packetSize) {
    //read the packet into a buffer-->
    int len = Udp.read(packetBuffer , UDP_TX_PACKET_MAX_SIZE); //read packet into packetBuffer array of max size UDP_TX_PACKET_MAX_SIZE
    if (len > 0) {
      packetBuffer[len] = 0;
    }
    String data(packetBuffer); //convert char array to string
    String res = "";
    String boardNo = split(data)[0];
    //Check if the incoming message is meant for this board?-->
    if(boardNo.toInt() == readInt(0)){
      data.remove(0 , boardNo.length()+1);
      Serial.println(data);
      //If "WiFiInit" is inside of received message, then that means user wants to initialize.
      if(data.indexOf("GetStat") >= 0){
        //read pin values
        int stat;
        String urlStat = "";
        for(int i=0; i < sizeof(pins)/sizeof(uint8_t) ; i++){
          stat = digitalRead(pins[i]); 
          urlStat += String(stat) + ";";
        }
        urlStat.remove(urlStat.length() - 1);
        res = "{\"temperature\":\"20.2\",\"dev_stat\":\"" + urlStat + "\"}";
        //Serial.println(res);
        //res = String(jsonChar);
      }
  
      Udp.beginPacket(Udp.remoteIP() , Udp.remotePort());
      char udpmsg[100];
      String msg = "Received " + res;
      msg.toCharArray(udpmsg , msg.length()+1);
      //Serial.println("Line right here : ");
      Udp.write(udpmsg);
      //Udp.write("Received");
      Udp.endPacket();
    }
  }
}
```

**Here is the stack trace:**
```
Exception (28):
epc1=0x40203148 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: cont 
sp: 3fff1f30 end: 3fff21e0 offset: 01a0

>>>stack>>>
3fff20d0:  00000034 00000000 3fff2b24 4020313e  
3fff20e0:  00000034 00000004 3fff2110 4010068c  
3fff20f0:  3ffe84fc 00000004 3ffee810 00000000  
3fff2100:  3ffe84fc 00000004 3ffee810 402023ff  
3fff2110:  65636552 64657669 74227b20 65706d65  
3fff2120:  75746172 3a226572 2e303222 222c2232  
3fff2130:  5f766564 74617473 30223a22 303b303b  
3fff2140:  7d22303b 00000000 3ffe84ab 3fff11b0  
3fff2150:  00001388 00000001 3ffe8841 402045f8  
3fff2160:  00000000 400042db 00000203 3fff11b0  
3fff2170:  40004b31 3fffbed4 0000003f 00000034  
3fff2180:  00000000 00000000 00000000 3fffbe04  
3fff2190:  0000000f 00000001 3fffbe54 0000002f  
3fff21a0:  0000002b 3fff46a4 0000000f 00000007  
3fff21b0:  00000000 00000000 00000001 3fff11b0  
3fff21c0:  3fffdad0 00000000 3fff11a9 402041bc  
3fff21d0:  feefeffe feefeffe 3fff11c0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
**Here here is the decoded stack trace:**

```
0x4020313e: UdpContext::_reserve(unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiUdp.cpp line 67
:  (inlined by) UdpContext::append(char const*, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\include/UdpContext.h line 237
:  (inlined by) WiFiUDP::write(unsigned char const*, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiUdp.cpp line 204
0x4010068c: free at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x402023ff: loop at C:\Users\user\Desktop\Codes\Arduino Codes\UDP_Messaging/UDP_Messaging.ino line 128 (discriminator 1)
0x402045f8: Print::write(unsigned char const*, unsigned int) at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x402041bc: loop_wrapper at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```

I have been trying to solve this problem for the longest time, but I just can't figure out the root cause.
Any help would be appreciated. _/\_

d-a-v, 2018-09-01:
Please retry with latest version, and add prints of `ESP.getFreeHeap()`.

devyte, 2018-09-01:
@d-a-v is correct. Release 2.3.0 is ancient, and the list of issues fixed is beyond long. Please try latest release, or latest git.
Having said that, at first glance from the incomplete code posted above, I already see several things that look suspicious.
The "C" in MCVE is for "Complete", and your sketch above is not. Please add the missing code to get it to build.

nikrath, 2018-09-02:
**SOLVED**
@d-a-v  @devyte  thanks a lot for the replies. 
I updated to the latest version and that added a bit of extra lines of information whenever the board reset. Also, I did as @d-a-v said and used print lines from ESP.getFreeHeap(). 
This way I was able to pinpoint that the reset was caused due to memory error.
To anyone else having this problem, you may be having some dangling pointers (unused pointers) in your code, in which case you need to free them manually (using either delete or delete[](for arrays))

closed 2018-09-02 by nikrath