esp8266/Arduino #5102: ESP8266 doesn't use local DNS table, and almost always makes query.

Deimos1994, 2018-09-01:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				Nodemcu v1.0
- Core Version:			V2.4.2
- Development Env:		Arduino IDE
- Operating System:		Windows 7

### Settings in IDE

- Module:  			Nodemcu|
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	       [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

ESP Always make DNS query 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>

ESP8266WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(19200);
  Serial.println();
  
  delay(1000);
  WiFi.mode(WIFI_STA);
  
   IPAddress ip(192, 168, 4, 213);
   IPAddress gateway(192, 168, 4, 1);
   IPAddress subnet(255, 255, 255, 0);
   IPAddress dnsAdrr(8, 8, 8, 8);
   WiFi.config(ip, gateway, subnet, dnsAdrr);
   
  WiFiMulti.addAP("MyNetwork", "MyPassword");

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    HTTPClient http;
    Serial.print("[HTTP] begin...\n");
    http.begin("http://api.thingspeak.com/gvhgvjustrandomhv"); //HTTP    
    int httpCode = http.GET();
    Serial.printf("[HTTP] GET... code: %d\n", httpCode);
    http.end();
  }

  delay(60000);
  
}

```

### Debug Messages

Traffic in Wireshark, filtered by IP and DNS/HTTP packets

[image]

As you can see it makes dns query every time, but it shouden't, right?
If I understood everything correctly ESP8266 should have a local dns table, that stores 3 addresses (when using lwip 2).
Of course if I replace api.thingspeak.com with its IP address 34.226.44.238 then no more DNS queries.
Why doesn't it use local DNS table?

**Serial**
Debug level: HTTP_CLIENT

```
lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc

scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with MyNetwork, channel 11
dhcp client start...
state: 5 -> 0 (0)
rm 0
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with W-Network, channel 11
ip:192.168.4.213,mask:255.255.255.0,gw:192.168.4.1
ip:192.168.4.213,mask:255.255.255.0,gw:192.168.4.1
[HTTP] begin...
[HTTP-Client][begin] url: http://api.thingspeak.com/gvhgvjustrandomhv
[HTTP-Client][begin] host: api.thingspeak.com port: 80 url: /gvhgvjustrandomhv
[HTTP-Client] connected to api.thingspeak.com:80
[HTTP-Client] sending request header
-----
GET /gvhgvjustrandomhv HTTP/1.1
Host: api.thingspeak.com
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=utf-8'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 728'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Status: 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'X-Request-Id: 0b80b791-a4a6-47c4-8ec1-2a960dcd8a6d'
[HTTP-Client][handleHeaderResponse] RX: 'X-Runtime: 0.003208'
[HTTP-Client][handleHeaderResponse] RX: 'X-Powered-By: Phusion Passenger 4.0.57'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Sat, 01 Sep 2018 21:24:11 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: nginx/1.9.3 + Phusion Passenger 4.0.57'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 404
[HTTP-Client][handleHeaderResponse] size: 728
[HTTP] GET... code: 404
[HTTP-Client][end] still data in buffer (728), clean up.
[HTTP-Client][end] tcp stop
pm open,type:2 0
[HTTP] begin...
[HTTP-Client][begin] url: http://api.thingspeak.com/gvhgvjustrandomhv
[HTTP-Client][begin] host: api.thingspeak.com port: 80 url: /gvhgvjustrandomhv
[HTTP-Client] connected to api.thingspeak.com:80
[HTTP-Client] sending request header
-----
GET /gvhgvjustrandomhv HTTP/1.1
Host: api.thingspeak.com
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=utf-8'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 728'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Status: 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'X-Request-Id: 4f64ec15-68f8-4dff-a8fe-db10ef64ca66'
[HTTP-Client][handleHeaderResponse] RX: 'X-Runtime: 0.004759'
[HTTP-Client][handleHeaderResponse] RX: 'X-Powered-By: Phusion Passenger 4.0.57'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Sat, 01 Sep 2018 21:25:12 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: nginx/1.9.3 + Phusion Passenger 4.0.57'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 404
[HTTP-Client][handleHeaderResponse] size: 728
[HTTP] GET... code: 404
[HTTP-Client][end] tcp stop
[HTTP] begin...
[HTTP-Client][begin] url: http://api.thingspeak.com/gvhgvjustrandomhv
[HTTP-Client][begin] host: api.thingspeak.com port: 80 url: /gvhgvjustrandomhv
[HTTP-Client] connected to api.thingspeak.com:80
[HTTP-Client] sending request header
-----
GET /gvhgvjustrandomhv HTTP/1.1
Host: api.thingspeak.com
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0

-----
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=utf-8'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 728'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Status: 404 Not Found'
[HTTP-Client][handleHeaderResponse] RX: 'X-Request-Id: 44ec1930-6ef0-4623-9a25-753917819ca4'
[HTTP-Client][handleHeaderResponse] RX: 'X-Runtime: 0.004188'
[HTTP-Client][handleHeaderResponse] RX: 'X-Powered-By: Phusion Passenger 4.0.57'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Sat, 01 Sep 2018 21:26:13 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: nginx/1.9.3 + Phusion Passenger 4.0.57'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 404
[HTTP-Client][handleHeaderResponse] size: 728
[HTTP] GET... code: 404
[HTTP-Client][end] still data in buffer (728), clean up.
[HTTP-Client][end] tcp stop

```

d-a-v, 2018-09-02:
You are right.
I guess debugging inside lwIP will answer to that question.

Deimos1994, 2018-09-02:
If only I knew how to do lwIP debugging...
By the way, if I use delay(10000) then it makes DNS queries after every 5 or 6 HTTP requests, but often it doesn't follow this rule...
[image]

Wireshark captured packets with delays 10s & 60s (just in case)
[delay_10s_and_60s.zip](https://github.com/esp8266/Arduino/files/2343201/delay_10s_and_60s.zip)

d-a-v, 2018-09-02:
It seems like lwIP's DNS cache lasts 1mn

d-a-v, 2018-09-04:
After reading src (easier than rfc), this is how dns works.
Answers contain a TTL which is honored by lwIP. The entry is flushed on expiration.
Usually ttls are a day. Thingspeak's is less than one minute. 
That's probably their server-load-sharing way.
```
$ dig +noauthority +noquestion +noadditional +nostats +ttlunits api.thingspeak.com
...
api.thingspeak.com.      0s     IN      A       34.202.98.102
...
api.thingspeak.com.     52s     IN      A       34.226.44.238
...
```

devyte, 2018-09-05:
@d-a-v is correct, the ttl depends on the domain being queried, and lwip honors that. I did a quick test on some domains and found very different ttls between them. Curiously, for cloud or api domains the ttls seem to be much lower, and that's the case for the thingspeak one in question. 
It happens that the queries were made once per minute, and the ttl is also one minute, so there was a dns lookup every time. That coincides with querying once every 10 seconds, which makes a dns lookup every 5-6 queries.
Closing.

closed 2018-09-05 by devyte