esp8266/Arduino #5104: setting SPI Slave mode

paultanner, 2018-09-03:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

GuaAck, 2018-10-29:
Hallo,

I had difficulties to connect an Arduino DUE with ESP8266 as SPI-Slave. DUE and ESP are configured with SPI-Mode=0.  I tried alle modes 0 to 3 on the DUE, without any success.  The screen shots (attached)  from an oscilloscope show the reason: MISO does not match with MODE 0, where the sampling occures with the rising edge and the data change at the falling edge of SCK. (With a slower Arduino UNO it works nevertheless, but not very reliable.)

My work around: The ESP provides the option to delay the MISO-Signal. I found the refering registers and added one line in the file hspi_slave.c: of the SPISlave-Library (to the ESP8266):
`
void hspi_slave_begin(uint8_t status_len, void * arg)
{
 ....
    SPI1CMD = SPIBUSY;
    SPI1C2=(0x2<<SPIC2MOSIDN_S) | (0x3<<SPIC2MISODN_S) | (0x7<<SPIC2MISODN_S); // Added 28.10.2018, works for SCK = 4 MHz
    ETS_SPI_INTR_ATTACH(_hspi_slave_isr_handler,arg);
...
}

The delay also for MOSI seems not reasonable, but I found in the ESP32 reference that it has to be 0x2 for MODE0, so I added it, but I found no influence.

With this change, the timing was perfect and the communication with the DUE works reliable.

[image]

GuaAck, 2018-10-30:
Hallo again,
ther is much better solution:
SPI1C2=(0x2<<SPIC2MOSIDN_S) | (0x1<<SPIC2MISODM_S); 
(Please take care of "N_S" and "M_S".);

this it works also with much lower frequencies.

GuaAck

devyte, 2018-10-30:
@GuaAck could you please explain in detail what you're doing, and why the second solution is better?

GuaAck, 2018-10-30:
Hallo,

to answer the question:

I tried to connect an EP8266 with my Arduino DUE, it fails. The program works with the UNO and the MEGA.  As a reason I found, that the timing of MISO does not match with SPI-MODE0. (Setting the MODE 1 - 3 in the master had no success.) The slower UNO and MEGA fread the correct bit, but the much faster DUE does not (See photos in the post above).  I found somewhere, that the ESP32 has a  SPI-module similar to that one in the ESP8266.  On base of the technical reference I inserted  the delay of the MISO of 8 cycles (MISODN=7) and it works for 4 MHz. But because it is a simple delay, I had doubts that it works for lower clock frequencies (it does not), but this was no problem for my project.

I have also contacted Jiri Bilek (https://github.com/JiriBilek/WiFiSpiESP), I use his library. We worked together to find out, what happens. Finally Jiri has had the idea to set MISODM=1 and to forget MISODN. I tried it, and it works perfectly and independant on the SCK-frequency, tested with 4 MHZ and 320 kHz.

I hope this helps,
GuaAck

devyte, 2018-10-31:
@GuaAck thank you, that certainly clears up the intent at high level. But I meant at low level, as in what does the line of code do? What registers are accessed, what do the bits mean? Like you said, the ESP's registers aren't really documented, and you seem to have investigated, so my request is for the details, so that they can be added e.g. as comments to a PR or inline with the code.
And speaking of PR, how about making one with your code change? This core is a community effort, and your contribution is highly welcome. It's only fair that your name shows up in the contribution history :)