esp8266/Arduino #5115: Code does not compile when compiling without optimisations -Og instead of optimize small -Os

jantje, 2018-09-09:
Hi
When trying to make a hardware debug platform.txt I bumped into the compilation problem below.
`'Building file: C:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_postmortem.c'
'Starting C compile'
"C:\eclipse\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IL:/test/runtime-EclipseApplication_eng/esp/ff/core" -c -Wall -Wextra   -DDEBUG -Og -ggdb -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DDEBUG_ESP_PORT=Serial  -DARDUINO=10802 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="\"ESP8266_GENERIC\"" -DLED_BUILTIN=9  -DESP8266   -I"C:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" -I"C:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\variants\generic" -I"C:\Users\jan\git\hardware\jantje\ESP8266\libraries\gdb" -MMD -MP -MF"core\core\core_esp8266_postmortem.c.d" -MT"core\core\core_esp8266_postmortem.c.o" -D__IN_ECLIPSE__=1 "C:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_postmortem.c"  -o  "core\core\core_esp8266_postmortem.c.o"
C:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_postmortem.c:83:23: error: istr causes a section type conflict with iram_read_byte
     static const char istr[] ICACHE_RAM_ATTR = (str); \
                       ^
C:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_postmortem.c:190:5: note: in expansion of macro 'ets_printf_P'
     ets_printf_P("<<<stack<<<\n");
     ^
C:\eclipse\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_postmortem.c:66:29: note: 'iram_read_byte' was declared here
 static char ICACHE_RAM_ATTR iram_read_byte (const char *addr) {
                             ^
core\core\subdir.mk:356: recipe for target 'core\core\core_esp8266_postmortem.c.o' failed
make: *** [core\core\core_esp8266_postmortem.c.o] Error 1
`
The problem lies into the fact that the define ICACHE_RAM_ATTR  is used 2 times for a different variable.
Apparently optimized for small optimizes this to 1 variable which makes this problem is not visible when using -Os. Obviously optimizing for debugging does not allow this.

This can be fixed by changing the line
    `static const char istr[] ICACHE_RAM_ATTR = (str); \`
to
    `static const char istr[] __attribute__((section(".iram.text.2"))) = (str); \`

I didn't want to make this a push request because I am not sure "the change is an acceptable solution" as yesterday I had not heard about something called
` __attribute__((section(XXX)))`

Best regards
Jantje

earlephilhower, 2018-09-10:
@jantje Give PR #5116 a try, it applies the same changes I did to fix this for PROGMEM variables to the section names of all ICACHE variables.

jjsuwa, 2018-09-10:
Hi again,

`#define ICACHE_FLASH_ATTR   __attribute__((section(".irom0.text")))
#define ICACHE_RAM_ATTR     __attribute__((section(".iram.text")))
#define ICACHE_RODATA_ATTR  __attribute__((section(".irom.text")))`

The original is of course OK, but

`#define ICACHE_FLASH_ATTR   __attribute__((section(".irom0.text.")))
#define ICACHE_RAM_ATTR     __attribute__((section(".iram.text.")))
#define ICACHE_RODATA_ATTR  __attribute__((section(".irom.text.")))`

is not. ;-)

earlephilhower, 2018-09-10:
I take it back, I read your comment wrong.  The version with the periods is already there and is similar to PROGMEM, so there's no problem with my PR as-is.

earlephilhower, 2018-09-10:
D'oh, too early, I see the problem you're pointing out is in the linker scripts, mybad!  Updated the PR.

earlephilhower, 2018-09-14:
Reopening since there seems to be some issue with the new linker sections and the patch is presently undone.

PR #5117 (2018-09-11): Make unique sections for ICACHE_* variables
Similar to PROGMEM changes, make the name of each ICACHE_* variable/fuction
unique to avoid issues with section conflicts.

Fixes #5115 

Added `.irom0.text.*` and `.iram.text.*` to ldscript.
commit 622569cd (2018-09-11): Make unique sections for ICACHE_* variables (#5117)

Similar to PROGMEM changes, make the name of each ICACHE_* variable/fuction
unique to avoid issues with section conflicts.

Fixes #5115
PR #5150 (2018-09-21): Move ICACHE_* to unique sections, local LD script
Similar to PROGMEM changes, make the name of each ICACHE_* variable/fuction
unique to avoid issues with section conflicts.

Also rename the generated LD linker script to avoid issue with older copies
of the eagle.app.v6.common.ld which were generated by the build process
in a global directory before being moved to the {build.path}.  The linker
would use the older, generated *.ld file instead of the generated one, which
would lead to runtime failures on some systems and cause the VTABLE location
to not correspond to the IDE menu selection.

Fixe [... cut]
commit 38779149 (2018-09-21): Move ICACHE_* to unique sections, local LD script (#5150)

* Move ICACHE_* to unique sections, local LD script

Similar to PROGMEM changes, make the name of each ICACHE_* variable/fuction
unique to avoid issues with section conflicts.

Also rename the generated LD linker script to avoid issue with older copies
of the eagle.app.v6.common.ld which were generated by the build process
in a global directory before being moved to the {build.path}.  The linker
would use the older, generated *.ld file instead of the generated one, which
would lead to runtime failures on some systems and cause the VTAB [... cut]
closed 2018-09-21 by devyte