esp8266/Arduino #5118: Function cont_get_free_stack is not available anymore

lmarmisa, 2018-09-10:
The new release 2.4.2 includes this change:

 . Move continuation stack from .bss onto sys stack, saving 4kB of RAM, if WPS is not needed (#4622)

But there is a side effect: the function **cont_get_free_stack(&g_cont)** [cont.h] does not work because the variable **g_cont** is not defined anymore (file core_esp8266_main.cpp).

My programs include stack statistics but I am unable to maintain this feature if I use the new release 2.4.2.

d-a-v, 2018-09-10:
Is it working with `g_pcont` instead of `&g_cont` ?

lmarmisa, 2018-09-11:
Yes, **g_pcont** did the trick. Thanks for your help.

I did not use the variable g_pcont before because this comment [core_esp8266_main.cpp] confused me:

/* Not static, **used in core_esp8266_postmortem.c**.
 * Placed into noinit section because we assign value to this variable
 * before .bss is zero-filled, and need to preserve the value.
 */
cont_t* g_pcont __attribute__((section(".noinit")));

devyte, 2018-09-13:
This is stupid. We need an interface for free stack in ESPClass as a wrapper to this.

devyte, 2018-09-13:
@lmarmisa please test #5133 .

PR #5133 (2018-09-13): implement EspClass::getFreeContStack method
Fixes #5118
closed 2018-09-13 by devyte