esp8266/Arduino #5123: boot-up problem after using Serial.swap() because of GPIO15 !

Rasoul-Khajavi, 2018-09-11:
Hi , i'm having a connection through UART and i use serial.swap() to change the pins of UART for having communication with another device . ( i cant use original pins because of problems with Serial programming whitch causes the 2nd device to crash )
after i use serial.swap() the 2 pins for UART are GPIO13 and GPIO15 . 
but we should pull-down GPIO15 for booting up from flash ... 
how can i use the second UART pins but also making sure that if my module resets it will boot up from flash (and not from SD-Card ) ?

5chufti, 2018-09-12:
and why do you think this is an issue with the esp srduino core?
Did you read and understand the questionair you were presented when starting the issue?
Please close the issue and discuss your problem on an appropriate user forum (e.g. www.esp8266.com)

devyte, 2018-09-12:
Closing. See issue POLICY doc and issue template instructions for more details.

closed 2018-09-12 by devyte