esp8266/Arduino #5139: HTTP client gets randomly stuck, with its stream's available() returning 0 almost forever

teo1978, 2018-09-15:
I have the following code (I have simplified it to post it here) that downloads a file from an URL and streams it to some physical output.

I read the downloaded data a few bytes at a time and use it as I download it beacause it's to big to store into the RAM (around 250 kB).

```
  uint8_t httpbuf[128] = {0};
  
  HTTPClient http;
  
  String url="http://some.url/";
  
  int httpcode=200;
  int len=0;
  
  http.begin(url);
  httpcode=http.GET();
  WiFiClient* httpstream = http.getStreamPtr();
  if (httpcode==200) {
    len=http.getSize();
  
    while (len > 0 || len ==-1) {
      int gotbytes=0;
    
      if (!http.connected()) {
        break;
      }
    
      size_t avl=httpstream->available();
      
      if (avl) {
        gotbytes = httpstream->readBytes(httpbuf, ((avl>sizeof(httpbuf))?sizeof(httpbuf):avl));
        Serial.println("Got some bytes");
      }
      else {
        Serial.println("No bytes available");
        delay(100); // can change this to any value, the result doesn't change.
      }
      if (gotbytes>0) {
        if (len>0) len=max(0, len-gotbytes);
        
        // use the data in httpbuf to generate some sort of physical output
        
      }
    }
    
    
  }
  else {
    // http error
  }
```

What happens is that often, randomly, `httpstream->available()` starts returning always 0 for very long periods of time.

This is not a network issue or a server issue, as I have never, ever observed anything like this on any kind of http client before.

The expected output would be lots of `Got some bytes` and some `No bytes available` from time to time, eventually resulting in downloading the whole file.

The observed result is either:
1) I start getting dozens of consecutive `No bytes available`, with some `Got some bytes` from time to time, so the download is extremely slow, but eventually succesfully finishes, or it suddenly "wakes up" and resume normal speed (i.e. I stop getting "No bytes available" and get mostly "got some bytes" until the end)
or
2) I start getting `No bytes available` forever, until eventually the connection gets closed, so `http.connected()` return false and breaks the loop, without finishing the download.

P.S. until this is fixed, suggestions for a workaround would be highly appreciated.

devyte, 2018-09-18:
@teo1978 when you opened this issue, you were presented with an issue template that contains specific instructions and requested info. You have ignored this.
I am keeping this open only because it rings a bell in my mind, though I can't place why yet.
Please edit your post, and make the sketch **Complete** (i.e.: it can be built and loaded directly on a board) and **Verifiable** (add a specific url that shows the problem). Please make sure to keep it **Minimal**, i.e.: do NOT post your entire project.

devyte, 2018-09-18:
It's not about what I do with the info, as this is not a service provided to users, but rather a community effort. The issue you encountered affects you, so you're the primary interested party for pursuing a fix. If you don't have the time to follow the minimum instructions in the issue template, then reporting this was just a waste of your own time, as well as mine.
I invite you again to complete the code above, so that someone else can take a look. If you're not willing or able, then please close this.

teo1978, 2018-09-18:
> It's not about what I do with the info, as this is not a service provided to users, but rather a community effort

Exactly, this is not a service provided to the user and I am not here asking for help. I am reporting an issue hoping this helps the developers (or any contributor) fix it and hence improve the library. I provided as much information as I have and spent as much effort as I am willing to spend. It may be useful or it may be not.

I don't _expect_ anybody to step in right away and start looking at the issue and fix it. I don't think closing the issue would be a sensible thing to do either, but that's up to the criteria of whoever maintains the project.

> The issue you encountered affects you, so you're the primary interested party for pursuing a fix

I am _an_ interested party; the abovementioned community, as well as anybody using the library, also are. Who is _more_ interested than anybody else, I don't know and I don't think is of any relevance.

> I invite you again to complete the code above

I'll probably do that when I get some time to: after all, as you said I am interested in getting this fixed, or even just in digging into it enough to find a workaround.

Or somebody else might also be willing to do that.

devyte, 2018-09-22:
@teo1978 there we go! That wasn't so hard now was it?
Now just please provide the core version (release version like 2.4.2 or git date/hash, etc),  and flash size + config (what size spiffs), and the required info should be close to complete.

devyte, 2018-09-22:
Notes:
-Stream O(n2)
-SPIFFS large file
-SPIFFS large size

teo1978, 2018-09-22:
> That wasn't so hard now was it?

Nope, which is why I din't think you needed _me_ to do it.

> Now just please provide...

Ok, will do tomorrow

> Notes:

I guess those notes are not for me, are they?

teo1978, 2018-09-23:
>  core version (release version like 2.4.2 or git date/hash, etc)

2.4.0, I got it via the Arduino IDE's Board Manager.

> and flash size + config (what size spiffs)

As I said it's this board:
https://learn.sparkfun.com/tutorials/esp8266-thing-development-board-hookup-guide?_ga=2.104455811.26954385.1537716697-914235417.1536518212

That's all I know.

d-a-v, 2018-09-23:
Would you please at least try 2.4.2 ?
Though git version is always better when reporting bugs.

teo1978, 2018-09-23:
Sorry, I have installed 2.4.2 now and I keep observing the same behavior. (do you? have you tried it??)

> Though git version is always better when reporting bugs.

I'd be happy to but I'm not sure how to do that, as I usually install the library through the Arduino IDE's Board Manager.

Not sure if I can just download the git version and simply overwrite the currently installed one with it?

Pablo2048, 2018-09-23:
Well the documentation is far from perfect, but did you read this https://github.com/esp8266/Arduino#using-git-version ?

d-a-v, 2018-09-24:
@teo1978 
This looks pretty much like #4176.
#5126 addresses this issue. We are looking for testers.
Once you have installed the git version, please try with this pull request.

### How to try

```
git fetch origin pull/5126/head:sack
git checkout sack
```

Revert to master:

```
git checkout master
```

In case of updates in the PR, do this before restarting the above.

```
git checkout master
git branch -D sack
```

teo1978, 2018-09-24:
Sorry, I intended to mark as duplicate of  #4176 (maybe I can't do that). I thought I would get that option (or not) and a confirmation prompt before closing.

Anyway, it does seem to be that issue. I am testing the PR and will report there.

Thank you.

closed 2018-09-24 by teo1978