esp8266/Arduino #5140: pwm works only for min and max values in core 2.4.2 - works great in 2.4.1

jplavoiemtl, 2018-09-16:
If you do a simple pwm analogWrite to a value beetween 0 or max=1023, the pwm output is not updated only in current release core 2.4.2.  If using 1023 or o for pwm the pwm output is properly set.  So only the intermediate values are affected by the bug.

Reverting back to core 2.4.1, it works fine the LED is properly varying intensity as per the proper pwm value.  The problem is very easy to see on my end.

I use a WeMos D1 R2 & mini board and I see that in core 2.4.2 the board is now defined as LOLIN also so maybe the problem is related to that?

Code is very simple and easy to reproduce and can be seen very easily on my LED.

`analogWrite(LED_Gpin, pwmVal)`

I see that my LED is properly dimmed in previous core 2.4.1 and is is completely off or on in core 2.4.2 and only respond to the min or max values.

earlephilhower, 2018-09-16:
Please post a MCVE sketch using LED_BUILTIN, as this is not reproducible by me and others have used PWM w/o incident.  The WEMOS->LOLIN was a name only change (I'm using that board, in fact) and didn't affect code.

jplavoiemtl, 2018-09-16:
Here is my code

`
#include <Timer.h>
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <PubSubClient.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <ESP8266mDNS.h>         //for OTA programming
#include <WiFiUdp.h>             //for OTA programming
#include <ArduinoOTA.h>          //for OTA programming

const int ledPin=2;              // Built-in Led on pin 2, active low
const int LED_Apin=5;            // Yel LED on GPIO5 or Wemos D1, active high
const int LED_Bpin=4;            // Red LED on GPIO4 or Wemos D2, active high
const int LED_Cpin=14;           // Yel LED on GPIO14 or Wemos D5, active high
const int LED_Dpin=12;           // Yel LED on GPIO12 or Wemos D6, active high
const int LED_Epin=13;           // Yel LED on GPIO13 or Wemos D7, active high
const int LED_Fpin=15;           // Yel LED on GPIO13 or Wemos D8, active high
const int LED_Gpin=0;           // Yel LED on GPIO0 or Wemos D3, active high

// Variables
int periodState = LOW;                    //Used to toggle High and Low flashes
bool first = true;
bool sequenceDone = true;                 //LED sequence done or occuring
int bit = 0;                              //bit counter for flashing sequence
String pgm_Str;                           //programming string for main loop in serial monitor
Timer t_LED;                              //Timer for LED flashing pwm mode
Timer t_Chaser;                           //Timer for LED flashing chaser mode
Timer t_Ring;                             //Timer for LED flashing ring mode

int ledEvent;
int chaserLedEvent;                       //for chaser timer
int ringLedEvent;                         //for ring timer
bool otaMode;                             //OTA mode flag
bool pgmMode = false;                     //PGM mode variable for main loop
bool chaserMode = false;                  //Chaser mode when true. When false: pwm mode.
bool ringMode = false;
bool drbellActive;                        //Doorbell active with flashing LED or inactive 
bool flasherDisable = false;              //Disable completely flasher and LEDs

int LED_A_Pattern;                        //16 bits active pattern for LED A
int LED_A_Pattern1;                       //16 bits first pattern for LED A
int LED_A_Pattern2;                       //16 bits second pattern for LED A
int LED_B_Pattern;                        //16 bits active pattern for LED B
int LED_B_Pattern1;                       //16 bits first pattern for LED B
int LED_B_Pattern2;                       //16 bits second pattern for LED B
int LED_C_Pattern;                        //16 bits active pattern for LED C
int LED_C_Pattern1;                       //16 bits first pattern for LED C
int LED_C_Pattern2;                       //16 bits second pattern for LED C
int LED_D_Pattern;                        //16 bits active pattern for LED D
int LED_D_Pattern1;                       //16 bits first pattern for LED D
int LED_D_Pattern2;                       //16 bits second pattern for LED D
int LED_E_Pattern;                        //16 bits active pattern for LED E
int LED_E_Pattern1;                       //16 bits first pattern for LED E
int LED_E_Pattern2;                       //16 bits second pattern for LED E
int LED_F_Pattern;                        //16 bits active pattern for LED F
int LED_F_Pattern1;                       //16 bits first pattern for LED F
int LED_F_Pattern2;                       //16 bits second pattern for LED F
int LED_G_Pattern;                        //16 bits active pattern for LED G
int LED_G_Pattern1;                       //16 bits first pattern for LED G
int LED_G_Pattern2;                       //16 bits second pattern for LED G

int LED_PatCounter;                       //Counter for active patterns for all LEDs
int patternNumber = 1;                    //used to enter LED patterns via serial port
int sequenceNum;                          //LED sequence counter for each animation
int maxSequence = 13;                     //Nombre de séquences d'animation
int pgmResponse;     
int pwmVal = 2;                           //pwm value default = 2 marchait, il faut maintenant 1023??
[... 868 lines cut ...]
     // Process message when new line character is recieved
      if (received == '\n') {
        pwmVal = pgm_Str.toInt();
//        analogWrite(LED_Apin, pwmVal);
//        analogWrite(LED_Bpin, pwmVal);
//        analogWrite(LED_Cpin, pwmVal);
//        analogWrite(LED_Dpin, pwmVal);
//        analogWrite(LED_Epin, pwmVal);
//        analogWrite(LED_Fpin, pwmVal);
//        analogWrite(LED_Gpin, pwmVal);        
        Serial.print("pwm = ");
        Serial.println(pwmVal, DEC);

        pgm_Str = "";
        Serial.println("Enter pwm 0-1023: ");
      }
    }
  }

  if(otaMode==1) {                    
    ArduinoOTA.handle();
  }

  client.loop();
}`

jplavoiemtl, 2018-09-16:
Sorry I don't have another development board and this one is in service in my home.   I cannot switch code or play with it easily like you asked unfortunately.  I reprogram the unit over the air for the tests I did to switch esp8266 core versions on my Arduino IDE.

The code I listed is too long I agree, but I didn't want to miss a part for you guys.  I simply do an analogWrite to a value between extrems to change LED brigthness so it is very simple.  No loop to vary the brightness or anything, just output a value. I don't vary pwm and any constant I put between 0 and 1023 do not give any output on the LED outputs with core 2.4.2.  I reverted to core 2.4.1 for now so that I'm able to dim my LEDs (they were either off or full on).  I thought this was easily to reproduce and hoped it could be resolved without too much difficulties.

earlephilhower, 2018-09-16:
The kind of code you've pasted is tested and verified working on its own, so there must be something else going on somewhere else in the code as a side effect.  I thought it might have been Ticker, which uses timers, which would be the key but that also runs just fine for me:
````
#include <Ticker.h>

Ticker ticktock;

int cnt = 0;
void cb() {
  cnt++;
  ticktock.attach(1,  cb);
}

void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  ticktock.attach(.1,  cb);
}

void loop() {
  for (int i=0; i<=1023; i++) { analogWrite(LED_BUILTIN, i); delay(1); }
  for (int i=1023; i>=0; i--) { analogWrite(LED_BUILTIN, i); delay(1); }
  Serial.printf("cb: %d\n", cnt);
}
````

earlephilhower, 2018-09-16:
Also I see you do an `analogWrite(0)` so I tried the following, and am also seeing it change intensity as expected:
````
#include <Ticker.h>

Ticker ticktock;

int cnt = 0;
void cb() {
  cnt++;
  ticktock.attach(1,  cb);
}

void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  ticktock.attach(.1,  cb);
}

void loop() {
//  for (int i=0; i<=1023; i++) { analogWrite(LED_BUILTIN, i); delay(1); }
//  for (int i=1023; i>=0; i--) { analogWrite(LED_BUILTIN, i); delay(1); }
  analogWrite(LED_BUILTIN, 0);
  analogWrite(LED_BUILTIN, 800);
  delay(2000);
  analogWrite(LED_BUILTIN, 0);
  analogWrite(LED_BUILTIN, 300);
  delay(2000);
  Serial.printf("cb: %d\n", cnt);
}
````

jplavoiemtl, 2018-09-16:
These are good tests indeed.  Like you say the problem must be a side effect from something else. I also do some digital writes on these outputs when I change mode.  

On a side note my project is actually a LED chaser with 7 LEDs flashing some patterns with digital writes.  When I put my module in "stealth" mode I use pwm to reduce the intensities of the LEDs and stop the LED chaser.  This is where the problem occurs.  Anyway I appreciate your efforts to reproduce the problem and I'm a bit sad because I don't have the proper hardware setup to make further tests like you asked.  For this reason if you suggest to close the issue, it's ok with me since I have a workaround to go back to 2.4.1.  Thanks.

devyte, 2018-09-17:
@jplavoiemtl I suggest you work to reduce your code to something minimal that reproduces the issue, or add debug statements at key positions to give you hints for what's going on, or get another board for testing.
Closing this due to lack of info and can't reproduce.

closed 2018-09-17 by devyte