esp8266/Arduino #5141: strptime() parses YYYYMMDD date incorrectly

bitmario, 2018-09-16:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12F
- Core Version:			2.4.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0
- Flash Mode:			dio
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- CPU Frequency:		80MHz
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

`strptime()` does not parse input correctly in cases where there are no literals between the conversion specifications.
It seems like the full value is parsed into the `tm` member corresponding to the first conversion specification (year in my case) and also mangles it somehow.

### Sketch

```cpp

#include <time.h>

void setup() {
  Serial.begin(115200);
  while (!Serial) {}
  
  struct tm tm = {0};
  char dateStr[] = "20180916";
  strptime(dateStr, "%Y%m%d", &tm);

  Serial.println(tm.tm_sec);  // 0
  Serial.println(tm.tm_min); // 0
  Serial.println(tm.tm_hour); // 0
  Serial.println(tm.tm_mday); // 0 - WRONG
  Serial.println(tm.tm_mon); // 0 - WRONG
  Serial.println(tm.tm_year); // 20179016 - WRONG
  Serial.println(tm.tm_wday); // 0
  Serial.println(tm.tm_yday); // 0 - WRONG
  Serial.println(tm.tm_isdst); // 0
}

void loop() {}

```

However, this works fine:

```cpp
  char dateStr[] = "2018-09-16";
  strptime(dateStr, "%Y-%m-%d", &tm);
```

### Comparison with GCC 8.1.0 on Linux

```cpp

#include <time.h>
#include <iostream>

int main()
{
    struct tm tm = {0};
    char dateStr[] = "20180916";
    strptime(dateStr, "%Y%m%d", &tm);

    std::cout << tm.tm_sec << std::endl;
    std::cout << tm.tm_min << std::endl;
    std::cout << tm.tm_hour << std::endl;
    std::cout << tm.tm_mday << std::endl;
    std::cout << tm.tm_mon << std::endl;
    std::cout << tm.tm_year << std::endl;
    std::cout << tm.tm_wday << std::endl;
    std::cout << tm.tm_yday << std::endl;
    std::cout << tm.tm_isdst << std::endl;

    return 0;
}

/*
OUTPUT:
0
0
0
16
8
118
0
258
0
*/

```

igrr, 2018-09-17:
In fact, strptime comes from Newlib. Latest version of which has the same behavior, just ran into the same a few days ago.

bitmario, 2018-09-17:
Thanks for the insight @devyte and @igrr. I was not aware that this was implemented externally.
I'll see if I can patch this upstream, or at least report it.