esp8266/Arduino #5166: Wrong documentation of WiFiServer::write

UlliBien, 2018-09-25:
In [https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/server-class.html](url) write point to [https://www.arduino.cc/en/Reference/WiFiServerWrite](url). There is written:
> Write data to all the clients connected to a server. 

That is not true for this library. The method is not implemented:
`size_t WiFiServer::write(const uint8_t *buffer, size_t size) {
    // write to all clients
    // not implemented
    (void) buffer;
    (void) size;
    return 0;
}`

d-a-v, 2018-09-25:
Is this function needed, or should we just update the doc ?

RudyFiero, 2018-09-28:
> Write data to all the clients connected to a server.

I can see a function like that as being useful but I do not think that should be the default behaviour. I wonder if the comment in that example is correct (for non-ESP8266 devices) or if it was a not quite write description. 

What if you had two clients connected to the server but loaded with different sub-pages? It can really make a mess of things. But then there is the case of the same web page sending updated status. 

`bool WebSocketsServer::broadcastBIN(uint8_t * payload, size_t length, bool headerToPayload) {`

This is useful, but it is a specific case, not the default.

UlliBien, 2018-09-28:
I think, it is not very useful. In deed, I did not expect the existence of such a function. 

But what ever you the dics should fit to the code.

devyte, 2018-09-29:
Some comments here:
- About whether it's useful or not, it's a matter of user application. Off the top of my head, I can think of a telnet example, where a programmed ESP reboot could be announced to all connected clients, like Linux does on its consoles. The use model is the server side initiates communication for some reason, and sends data to all connected clients.
- Whether or not it's useful, Arduino specifies it, and we should aim for compatibility.
- We're talking about TCP server sockets, not about websockets or the webserver.
- l seem to remember seeing this before. Did you search the tracker? Could be no longer open, or a PR instead of issue.

yoursunny, 2018-10-02:
> l seem to remember seeing this before. Did you search the tracker? Could be no longer open, or a PR instead of issue.

It's #2743.

earlephilhower, 2019-07-25:
Closed via #6338

closed 2019-07-25 by earlephilhower