esp8266/Arduino #5169: Unable to cache built core, please tell esp8266 maintainers to follow http://goo.gl/QdCUjo

philbowles, 2018-09-26:
- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:				all
- Core Version:			2.4.2
- Development Env:		Arduino IDE 1.8.7
- Operating System:		Windows 10

### Settings in IDE

N/A happens on every occasion , multiple boards, speeds etc

### Problem Description

Since installing Arduino IDE 1.8.7 I notice the following message in all builds. 
"Unable to cache built core, please tell esp8266 maintainers to follow http://goo.gl/QdCUjo"
The builds works fine, nothing else seems affected, just wondering a) if you are aware b) what it means

```...
"C:\\Users\\phil\\AppData\\Local\\Temp\\arduino_build_567284/arduino.ar" "C:\\Users\\phil\\AppData\\Local\\Temp\\arduino_build_567284\\core\\umm_malloc\\umm_malloc.c.o"
**_Unable to cache built core, please tell esp8266 maintainers to follow http://goo.gl/QdCUjo_**
Linking everything together...
"C:\\Users\\phil\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "C:\\Users\\phil\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\\Users\\phil\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.4.2/tools/sdk/ld/eagle.app.v6.common.ld"
...
```

5chufti, 2018-09-26:
Does the same problem (message) occour with the git version?

per1234, 2018-09-27:
This was fixed by https://github.com/esp8266/Arduino/commit/85e68093e94b4e5c5e711888b8b5facee1bd4f9f#diff-d638a47d1ca0e2a5585fb34c7d7fb0f9L95

devyte, 2018-10-01:
Closing per previous comment.

closed 2018-10-01 by devyte

ElFishi, 2018-10-04:
I have the same problem, but I have no clue how to fix it with the information given here. Can you help me out? I'm using the Arduino IDE 1.8.7 under Windows 10

per1234, 2018-10-04:
@ElFishi just ignore that message. This won't cause any problems for you other than a slower compilation time. The purpose of the message is to prompt hardware package authors to make the change needed to support caching. That has been accomplished here already. The warning will be fixed in the next release of the ESP8266 core for Arduino.

ElFishi, 2018-10-05:
Thanks fort responding, but it does cause a problem for me. I am missing the .ino.bin file that I use for a web update for my remote ESP8266. 
With IDE 1.8.4 the file is generated, with 1.8.7 it isn't.

per1234, 2018-10-05:
I can't reproduce that and I don't see what makes you think that problem is related to the core not being archived. Using Arduino IDE 1.8.7 and ESP8266 Core For Arduino 2.4.2, I get a .ino.bin file in the temporary build folder and I also get one in the sketch folder when I do **Sketch > Export Compiled Binary**. It seems to be off topic here.