esp8266/Arduino #5173: ESP8266WebServer responds sluggy to some client after commit 83a8076db87b77de6a7b500c18b85057e38ed117

jjsuwa, 2018-09-27:
**Example:**
HTTP server (esp8266/Arduino simple HTTP server w/mDNS):
```C++
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>

#define WIFI_SSID     "YOUR-WIFI-SSID"
#define WIFI_PASSWORD "YOUR-WIFI-PASSWORD"
#define MDNS_HOSTNAME "avahi"

static ESP8266WebServer webServer(80);

void setup() {
  Serial.begin(115200);
  delay(250);
  Serial.print(F("\n\nconnecting to WiFi"));
  WiFi.persistent(false);
  // WiFiClient::setDefaultNoDelay(false);
  // WiFiClient::setDefaultSync(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(String(F(WIFI_SSID)).c_str(), String(F(WIFI_PASSWORD)).c_str());
  for (; ; delay(500)) {
    if (WiFi.status() == WL_CONNECTED)
      break;
    Serial.print('.');
  }
  Serial.print(F(", done.\n"));  MDNS.begin(String(F(MDNS_HOSTNAME)).c_str());
  MDNS.addService(F("http"), F("tcp"), 80);
  webServer.onNotFound([]() {
    webServer.send(404);
  });
  webServer.on(F("/"), HTTP_GET, []() {
    webServer.sendHeader(F("Location"), String(F("http://" MDNS_HOSTNAME ".local/index.html")));
    webServer.send(302);
  });
  webServer.on(F("/index.html"), HTTP_GET, []() {
    webServer.send(200, F("text/html"), F("<!DOCTYPE html><html>"
                                          "<head><meta http-equiv=\"Content-Type\" content=\"text/html; charset=utf-8\"/><title>WiFi HTTP server test</title></head>"
                                          "<body><h1>WiFi HTTP server test</h1><p>hello world</p></body>"
                                          "</html>"));
  });
  webServer.begin();
}

void loop() {
  webServer.handleClient();
}
```
and 2 equivalent HTTP clients... (C# Windows 7/ndp20):
```C#
using System;
using System.Diagnostics;
using System.IO;
using System.Net;

namespace httprequest2avahi
{
    internal static class Program
    {
        public static void Main(String[] args)
        {
            String url = String.Format("http://{0}/index.html", Dns.GetHostAddresses("avahi.local")[0]);
            Stopwatch sw = new Stopwatch();
            for(Int32 i = 1; ; ++i)
            {
                HttpWebRequest req = (HttpWebRequest)WebRequest.Create(url);
                req.Timeout = 60000;
                req.ReadWriteTimeout = 60000;
                req.KeepAlive = false;
                req.Pipelined = false;
                sw.Reset();
                sw.Start();
                using(HttpWebResponse res = (HttpWebResponse)req.GetResponse())
                {
                    sw.Stop();
                    Double d = sw.Elapsed.TotalSeconds;
                    if(d > 1)
                        Console.WriteLine("[{0}] req.GetResponse(): {1} seconds", DateTime.Now, d);
                    if(res.StatusCode != HttpStatusCode.OK)
                        break;
                    using(Stream s = res.GetResponseStream())
                    using(StreamReader sr = new StreamReader(s))
                        sr.ReadToEnd();
                }
                Console.Write("connected {0} times\r", i);
            }
        }
    }
}
```
 (Python3 Windows 7 or Raspberry Pi 3B):
```Python
#!/usr/bin/env python3

import http.client
import itertools
import socket
from datetime import datetime

if __name__ == "__main__":
    conn = http.client.HTTPConnection(socket.gethostbyname("avahi.local"), 80, timeout=60)
    for i in itertools.count(1):
        t = datetime.now()
        conn.request("GET", "/index.html")
        d = (datetime.now() - t).total_seconds()
        if d > 1:
            print(" [{}] conn.request(): {} seconds\n".format(t, d))
        with conn.getresponse() as res:
            if res.status != 200:
                break
            res.read()
        print("connected {} times\r".format(i), end="")
```
On commit 88bd26bd74882e02c1dc0f0d529209a892849e3c (or earlier),
- both 2 abovementioned clients receive responses well (100+ per second)

However on commit 83a8076db87b77de6a7b500c18b85057e38ed117 ~ 7c91eee26d84ab454ae841271ba4cbd81a75360b,
- C# client receives suddenly slow (2~3 per second), but Pythons well
- `WiFiClient::setDefaultNoDelay(true)` recovers the performance

Moreover on commit 775eb9b34375daff2e301886815ddf91f3416df6,
- C# client performance degradation like the above
- `WiFiClient::setDefaultNoDelay(true)` is ineffective now

d-a-v, 2018-09-27:
I will not be able to run C# at the moment,
can you try this in libraries/ESP8266WiFi/src/include/ClientContext.h L493:
change
`uint8_t flags = TCP_WRITE_FLAG_MORE; // do not tcp-PuSH`
to
`uint8_t flags = 0; // TCP_WRITE_FLAG_MORE; // do not tcp-PuSH`

jjsuwa, 2018-09-27:
@d-a-v - can you try this ~

Bingo! It recovers C# client performance w/o `setDefaultNoDelay(true)`, and python one also works well :)

d-a-v, 2018-09-27:
Thanks !
Let me some time to try to set the PuSH flag only in the last write, as stated in this interesting [explanation](https://lists.gnu.org/archive/html/lwip-users/2009-11/msg00018.html).

d-a-v, 2018-09-27:
Can you please retry with C# and this:
```
            uint8_t flags = 0;
            if (next_chunk_size < _datasource->available())
                flags |= TCP_WRITE_FLAG_MORE; // do not tcp-PuSH
```

jjsuwa, 2018-09-27:
> @d-a-v - Can you please retry with C# and this

LGTM.