esp8266/Arduino #5175: Downloading large files to spiffs via https

MarcFinns, 2018-09-27:
I am observing at behaviour that I would like to confirm being as expected.
I need to download a file from an HTTPS service into SPIFFS. This file is fairly large and  when downloading using HTTP it is not a problem. This is what I have done in other projects for example with the json streaming parser. It looks like HTTPS is instead loading the whole thing in RAM, and then it crashes.
Is this what others are experiencing/ the expected behaviour? Is there a way to do it without lording it in RAM? ?
Thank you.

MarcFinns, 2018-09-27:
Apologies. Here it is.

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu 1.0]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[20Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [other] (serial upload only)

### Problem Description

I need to download a file from an HTTPS service directly into SPIFFS, as I used to do successfully with plain HTTP with files larger than the available RAM (e.g. json files).
The sketch shows about 30K free heap before the download starts. The download fails if the file is above 8k or so. The sketch gets stuck, no stack dump. Enabling the serial debug shows a ram issue (ssl->need_bytes=10464 > 6859).
It seems the file is loaded first into RAM and does not go directly to SPIFFS. Is this the expected behaviour? Is there a way to download it without copying it into RAM?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#define FS_NO_GLOBALS
#include <FS.h>

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <Arduino.h>

char* myssid = "GuestNet";
char* mypass = "xxxxx";

WiFiClientSecure client;

void setup()
{

  // Use some ram to simulate the full sketch utilisation
  int kb = 6;
  int *myArr = ( int* ) malloc( sizeof( int ) * kb * 500 );
  myArr[ 500 * kb - 1 ] = 4;

  Serial.begin(115200);
  delay(1000);

  Serial.println("START - Free Heap: " + String(ESP.getFreeHeap()));
  SPIFFS.begin();
  //SPIFFS.format();

  // Connect to WiFi
  Serial.print("Connecting to ");
  Serial.println(myssid);

  WiFi.mode(WIFI_STA);
[... 280 lines cut ...]
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 11872'
[HTTP-Client][handleHeaderResponse] RX: 'Cache-Control: public, max-age=900'
[HTTP-Client][handleHeaderResponse] RX: 'Expires: Thu, 27 Sep 2018 18:58:13 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Thu, 27 Sep 2018 18:43:13 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Max-Age: 86400'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Allow-Credentials: false'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Allow-Headers: Origin, X-Requested-With, Content-Type, Accept'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Allow-Methods: GET,POST,PUT'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Allow-Origin: *'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 11872
[HTTP] GET DONE with code 200
-- >> OPENING FILE...
-[HTTP] GET... code: 200
Free Heap: 10504
HTTP SIZE IS 11872
:rd 5, 1608, 0
:rdi 536, 5
ssl->need_bytes=10464 > 6859
:oom(11264)@?
failed to grow plain buffer
:rch 1608, 363
pm open,type:2 0

d-a-v, 2018-09-28:
>  ssl->need_bytes=10464 > 6859
:oom(11264)@?

This is an AxTLS known bug. You should try BearSSL.

Check [this](https://github.com/esp8266/Arduino/issues/2771#issuecomment-422709732).

MarcFinns, 2018-09-28:
Hi d-a-v, thanks for the hint.
I did implement it and in fact i confirm it works!
**HOWEVER** i see a behaviour that i would like to confirm is as expected and not a memory leak.
When the line: _BearSSL::WiFiClientSecure client;_ is executed, heap goes down by about 5K.
These 5K are never deallocated, even if the class that downloaded is long gone.
To me this seems not right.

devyte, 2018-09-29:
CC @earlephilhower

earlephilhower, 2018-09-29:
@MarcFinns that's a feature, not a bug. :)

~4.5K are taken from the heap to use as a 2nd stack as long as there is at least one `BearSSL::WiFi{Client,Server}Secure`.  If you have a global WiFiClientSecure, it will obviously never go away so you'll end up with that chunk never freeing.  I'd recommend a dynamically allocated object (i.e. `new ...`) or one that's stack-local inside a function that exits, to ensure it's deleted.  The 4.5K is constant no matter how many objects you've got, so you could have, say, 20 `BearSSL::WiFiClientSecure`s and it would still be 4.5K.

If you've got something like
````
BearSSL::WiFiClientSecure *client = new BearSSL::WiFiClientSecure;
...
(do some work)
...
delete client;
````
and still seeing the heap used, that's a bug and I'll need a MCVE to reproduce and debug.

MarcFinns, 2018-09-29:
It seems a bug. You get the behavior with both types of allocation and it can be reproduced with 2 lines of code - see below. Is this sufficient to report the bug?

#include <ESP8266WiFi.h>
void setup()
{
  Serial.begin(115200);
  delay(1000);
  Serial.println("Setup - before instantiation - Free Heap: " + String(ESP.getFreeHeap()));
  BearSSL::WiFiClientSecure *client = new BearSSL::WiFiClientSecure;
  client->stop();
  Serial.println("Setup - After instantiation - Free Heap: " + String(ESP.getFreeHeap()));
  delete client;
  Serial.println("Setup - After delete - Free Heap: " + String(ESP.getFreeHeap()));
}

void loop()
{
  Serial.println("Loop - Free residual Heap: " + String(ESP.getFreeHeap()));
  while (1)
    delay(1000);
}

earlephilhower, 2018-09-29:
Thanks, that'll do.  Appreciate the check!  I'll look into it.

ZinggJM, 2018-10-22:
@MarcFinns,

large file to Spiffs works for you with BearSSL? up to what size?

It does not work for me, issue #4814.

MarcFinns, 2018-10-22:
I used it to download every half hour 24 images of a weather animation. In the web service I can specify the jpg size and I now use 240x200. This means images are 10-25k depending on the cloud patterns. It would not work with axTLS but it works nicely with BearSSL. 
Please note that with bear you can specify the size of the receiving buffer and read from network/write to spiffs in small chunks. 
How large are your files?

ZinggJM, 2018-10-22:
Sorry, I found out in the meantime that my GxEPD2_32_Spiffs_Loader example works with BearSSL.
I still have an issue, e.g. with GxEPD_WiFi_Example, when I use it with an e-paper that needs a 15k graphics buffer. I have seen an issue that BearSSL takes up to 48k from heap.

MarcFinns, 2018-10-22:
BearSSL works with less than that. Have you tried reducing the buffers, not checking the fingerprint and reading the file in chunks?
see example below, hope it helps. let me know.
`
int getForecastImage(String host, String resource, String filename)
{

  int contentLength = -1;
  int httpCode;

  // HTTPS but dont verify certificates
  BearSSL::WiFiClientSecure client;
  client.setBufferSizes(1024, 256);
  client.setInsecure();

  // Connect
  client.connect(host, 443);

  // If not connected, return
  if (!client.connected())
  {
    client.stop();
    errLog("HTTPS: Can't connect");
    return -1;
  }

  // HTTP GET
  client.print(F("GET "));
  client.print(resource);
  client.print(F(" HTTP/1.1\r\nHost: "));
  client.print(host);
  client.print(F("\r\nUser-Agent: ESP8266\r\n"));
  client.print(F("\r\n"));

  // Handle headers
  while (client.connected())
  {
    String header = client.readStringUntil('\n');
    if (header.startsWith(F("HTTP/1.")))
    {
      httpCode = header.substring(9, 12).toInt();

      if (httpCode != 200)
      {
        errLog(String(F("HTTP GET code=")) + String(httpCode));
        client.stop();
        return -1;
      }
    }

    if (header.startsWith(F("Content-Length: ")))
    {
      contentLength = header.substring(15).toInt();
    }
    if (header == F("\r"))
    {
      break;
    }
  }

  if (!(contentLength > 0))
  {
    errLog(F("HTTP content length=0"));
    client.stop();
    return -1;
  }

  // Open file for write
  fs::File f = SPIFFS.open(filename, "w+");
  if (!f)
  {
    errLog( F("file open failed"));
    client.stop();
    return -1;
  }

  // Download file
  int remaining = contentLength;
  int received;
  uint8_t buff[512] = { 0 };

  // read all data from server
  while (client.available() && remaining > 0)
  {
    // read up to buffer size
    received = client.readBytes(buff, ((remaining > sizeof(buff)) ? sizeof(buff) : remaining));

    // write it to file
    f.write(buff, received);

    if (remaining > 0)
    {
      remaining -= received;
    }
    yield();
  }

  if (remaining != 0)
    errLog("[HTTP] Img truncated -" + String(remaining));

  // Close SPIFFS file
  f.close();

  // Stop client
  client.stop();
  return (remaining == 0 ? contentLength : -1);
}
`

ZinggJM, 2018-10-23:
Thank you for your response.
The site I download from does not support MFLN, returns false,

I tried to update my MCVE sketch to use with BearSSL to show the issue. **But my MCVE sketch works.**
So I need to analyze my issue with download to draw on e-paper from start.

devyte, 2018-10-23:
@ZinggJM this thread has gone in several directions already, and is getting a bit long. Given your previous comment, I'm closing it.
If after your analysis you think there is still a problem, please open a new issue, fill out the template, and provide a fresh explanation and MCVE.

closed 2018-10-23 by devyte

ZinggJM, 2018-10-24:
I had used this method (from class Client):
`        virtual int read(uint8_t *buf, size_t size) = 0;
`
If I use single byte reads instead, checking for available, it works.
`        virtual int available() = 0;
        virtual int read() = 0;
`

massgl, 2021-03-26:
Hi MarcFinns,

Thanks for sharing your work!
I am pretty new to coding in Arduino, but I have done some small hobby-projects inspired form various inputs around the internet.
I was really inspired by your work and am now trying to download a small jpg-file the my ESP8266. Hope it's OK, that I am using your code. However, I can't seem to make it work properly.

I am trying to download a jpg-file from the url: http://www.columbia.edu/~fdc/picture-of-something.jpg (first and best http:// site, I could find) ans name it image1.jpg

I am using Arduino IDE to code and using your code as below, where I deleted the fingerprints:

---------------------------------------
#define FS_NO_GLOBALS
#include <FS.h>

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <Arduino.h>

char* myssid = "Mass";
char* mypass = "remember";

WiFiClientSecure client;

void setup()
{

  // Use some ram to simulate the full sketch utilisation
  int kb = 6;
  int *myArr = ( int* ) malloc( sizeof( int ) * kb * 500 );
  myArr[ 500 * kb - 1 ] = 4;

  Serial.begin(115200);
  delay(1000);

  Serial.println("START - Free Heap: " + String(ESP.getFreeHeap()));
  SPIFFS.begin();
  //SPIFFS.format();

  // Connect to WiFi
  Serial.print("Connecting to ");
  Serial.println(myssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(myssid, mypass);

  // Wait for connection...
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(".");
  Serial.println("Connected to network " + String(WiFi.SSID()) + " with address " + String(WiFi.localIP().toString()));
}

void loop()
{

  // Download file
  downloadImage("http://www.columbia.edu/~fdc/picture-of-something.jpg",
                "/image1.jpg" );

  Serial.println("SPIFFS dir listing:");

  String fileName;
  fs::Dir dir = SPIFFS.openDir(F("/"));
  while (dir.next())
  {
    fileName = dir.fileName();

    Serial.println(fileName);
  }

  Serial.println( " -----  ALL DONE ----- ");
  while (true)
    delay(1000);
}

bool downloadImage(String url, String filename)
{

  HTTPClient http;
  bool outcome = true;

  // configure server and url
  http.begin(url);

  // start connection and send HTTP header
  int httpCode = http.GET();

  Serial.println(("[HTTP] GET DONE with code " + String(httpCode)));

  if (httpCode > 0)
  {
    Serial.println(F("-- >> OPENING FILE..."));

    SPIFFS.remove(filename);
    fs::File f = SPIFFS.open(filename, "w+");
    if (!f)
    {
      Serial.println(F("file open failed"));
      return false;
    }

    // HTTP header has been sent and Server response header has been handled
    Serial.printf("-[HTTP] GET... code: %d\n", httpCode);
    Serial.println("Free Heap: " + String(ESP.getFreeHeap()));

    // file found at server
    if (httpCode == HTTP_CODE_OK)
    {
      // get lenght of document (is -1 when Server sends no Content-Length header)
      int total = http.getSize();
      int len = total;

      Serial.println("HTTP SIZE IS " + String(total));

      // create buffer for read
      uint8_t buff[128] = { 0 };

      // get tcp stream
      WiFiClient * stream = http.getStreamPtr();

      // read all data from server
      while (http.connected() && (len > 0 || len == -1))
      {
        // get available data size
        size_t size = stream->available();

        if (size)
        {
          // read up to 128 byte
          int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));

          // write it to
          // f.write(buff, c);

          if (len > 0)
          {
            len -= c;
          }
        }
        delay(1);
      }

      Serial.println("[HTTP] connection closed or file end.");
    }
    f.close();
  }
  else
  {
    Serial.println("[HTTP] GET... failed, error: " + http.errorToString(httpCode));
    outcome = false;
  }
  http.end();
  return outcome;
}
---------------------------------------

The code is uploaded to the NodeMCU and looks like the jpg-file is downloaded correct with response code 200. Se below print from the Arduino IDE:

16:58:40.455 -> Connecting to Mass
16:58:41.011 -> ..............
16:58:47.805 -> Connected to network Mass with address 192.168.0.33
16:58:48.440 -> [HTTP] GET DONE with code 200
16:58:48.440 -> -- >> OPENING FILE...
16:58:48.549 -> -[HTTP] GET... code: 200
16:58:48.549 -> Free Heap: 27904
16:58:48.549 -> HTTP SIZE IS 44566
16:58:51.404 -> [HTTP] connection closed or file end.
16:58:51.404 -> SPIFFS dir listing:
16:58:51.404 -> /image1.jpg
16:58:51.439 ->  -----  ALL DONE ----- 

However, if I install FSBrowser (from Arduino IDE examples) and try to download the image from the SPIFFS, I get an image file containing 0 bytes. Please see the IDE-feedback below:

17:15:34.203 -> Filesystem initialized.
17:15:34.203 -> List of files at root of filesystem:
17:15:34.203 -> /image1.jpg (0b)
17:15:34.240 -> 
17:15:34.240 -> Connecting to Mass
17:15:34.240 -> scandone
17:15:34.389 -> scandone
17:15:34.389 -> state: 0 -> 2 (b0)
17:15:34.389 -> state: 2 -> 3 (0)
17:15:34.389 -> state: 3 -> 5 (10)
17:15:34.389 -> add 0
17:15:34.389 -> aid 3
17:15:34.389 -> cnt 
17:15:34.428 -> 
17:15:34.428 -> connected with Mass, channel 3
17:15:34.428 -> dhcp client start...
17:15:34.741 -> .....ip:192.168.0.33,mask:255.255.255.0,gw:192.168.0.1
17:15:37.257 -> .
17:15:37.257 -> Connected! IP address: 192.168.0.33
17:15:37.257 -> Open http://fsbrowser.local/edit to open the FileSystem Browser
17:15:37.257 -> HTTP server started
17:15:44.382 -> pm open,type:2 0

I am not able to find the reason why I get a 0 byte jpg-file downloaded and It drives me crazy!

If you find the time, I would really appreciate a hint to why the file bytes are lost during this process.

Thanks a lot in advance,
All the best,
Mass

massgl, 2021-03-26:
@MarcFinns 

Sorry for the bad formating. 

:)