esp8266/Arduino #5179: DNSServer can leak memory if dynamically allocated/free'ed

Makuna, 2018-09-28:
### Basic Infos
DNSServer does not have a destructor which can cause memory leaks if dynamically allocated.

### Problem Description

DNSServer needs a destructor.  Currently you can destruct the object and leave the internal `_buffer` memory still allocated causing a memory leak.  

This can be worked around by calling `stop()` before destructing, but this is an unsafe assumption.

earlephilhower, 2018-09-28:
I'm not seeing any leak potential.

`_buffer` is only `malloc()`'d  in `DNSServer::processNextRequest()`
https://github.com/esp8266/Arduino/blob/6314093fe59c1411c8c6c92a203aa307b0f2cc4d/libraries/DNSServer/src/DNSServer.cpp#L55
It's `free()`'d inside the same `if {...}` statement before any potential return from the method:
https://github.com/esp8266/Arduino/blob/6314093fe59c1411c8c6c92a203aa307b0f2cc4d/libraries/DNSServer/src/DNSServer.cpp#L73

If you've got the time, PRs for things like keywords.txt are always appreciated, too.

Makuna, 2018-09-28:
Wow, I looked deeper, and you are correct.  The code in the destructor should never be needed then and should be removed as it implies a use that is not present.

Further, there should not be a member at all, since it has a limited lifetime, it should just be passed into the methods that will need it rather than store an extra pointer for the lifetime of the object to just avoid passing an argument.

devyte, 2018-09-28:
@Makuna there's also my [ESPAsyncDNSServer](https://github.com/devyte/ESPAsyncDNSServer) in case you're interested. I only mention it because I've received little feedback, andand I'd like some :p

devyte, 2018-09-30:
@Makuna you self-assigned this. Do you intend to get it done within the next few days?

Makuna, 2018-09-30:
https://github.com/esp8266/Arduino/pull/5192

commit 656bf146 (2018-10-04): Dns server cleanup (#5194)

Clean up the DNSServer class.

Removed member variables that are not required outside a
member call lifetime, and add destructor/checks.

Fixes #5179
closed 2018-10-04 by earlephilhower