esp8266/Arduino #5187: There is no such word as "authentification"

jayeye, 2018-09-30:
The correct word is "authentication".

dragondaud, 2018-09-30:
@jayeye It really helps to include some details when opening an issue, however a quick grep finds:
```
./libraries/ESP8266WebServer/examples/SimpleAuthentification/SimpleAuthentification.ino:bool is_authentified() {
./libraries/ESP8266WebServer/examples/SimpleAuthentification/SimpleAuthentification.ino:  Serial.println("Enter is_authentified");
./libraries/ESP8266WebServer/examples/SimpleAuthentification/SimpleAuthentification.ino://root page can be accessed only if authentification is ok
./libraries/ESP8266WebServer/examples/SimpleAuthentification/SimpleAuthentification.ino:  if (!is_authentified()) {
./libraries/ESP8266WebServer/examples/SimpleAuthentification/SimpleAuthentification.ino://no need authentification
./libraries/ESP8266WebServer/examples/SimpleAuthentification/SimpleAuthentification.ino:    server.send(200, "text/plain", "this works without need of authentification");

```

The name of the example, and its directory, and several uses in code should be fixed, as well as well as 'is_authentified' should be 'is_authenticated'. Surprised this wasn't noticed earlier.

5chufti, 2018-09-30:
> The correct word is "authentication".

Says who?
https://en.oxforddictionaries.com/definition/authentification

igrr, 2018-09-30:
https://books.google.com/ngrams/graph?content=authentification%2Cauthentication&case_insensitive=on&year_start=1800&year_end=2008&corpus=15&smoothing=3&share=&direct_url=t4%3B%2Cauthentification%3B%2Cc0%3B%2Cs0%3B%3Bauthentification%3B%2Cc0%3B%3BAuthentification%3B%2Cc0%3B.t4%3B%2Cauthentication%3B%2Cc0%3B%2Cs0%3B%3Bauthentication%3B%2Cc0%3B%3BAuthentication%3B%2Cc0%3B%3BAUTHENTICATION%3B%2Cc0

>

closed 2018-09-30 by devyte