esp8266/Arduino #5207: Send dato MQ

smalhao, 2018-10-04:
When I run the code the serial monitor shows the value of the temperature but then it does not send to the MQTT broker nor to the database, if it runs the code in separate(in a file for each one), that is, one at a time it already sends or to the database or to the MQTT broker, how can I send both sides at the same time?
In my void loop() i have an error but i'm not see the error, my code is: 

   #include <ESP8266WiFi.h>
    #include <PubSubClient.h>
    #include <MySQL_Connection.h>
    #include <MySQL_Cursor.h>

    const char * ssid = "STARNAUTE4"; // network ssid wifi
    const char * password = "602af20122f8"; // wifi password network
    const char * mqttServer = "192.168.43.37"; // IP address of Raspberry Pi
    const int mqttPort = 1883; // Port by default of MQTT
    const char * mqttUser = "test"; // MQTT Username, if the authentication in the broker is configured
    const char * mqttPassword = "test"; // MQTT Password, if the authentication in the broker is configured
    IPAddress server_addr (192, 168, 43, 175); // IP of the MySQL server
    char login_username [] = "root"; // MySQL user login username
    char login_password [] = "admin"; // MySQL user login password

    float tempC // Define tempC variable by converting it to float data type.
    float temp_C;
    int reading; // Define reading variable by converting it from int data type;
    int read_value;
    float referenceVoltage; // Defines referenceVoltage variable by converting it to the float data type.
    int tempPin = A0; // Analog pin where it is connected or LM35
    String convert; // Initialize the convert variable as String
    char array [6]; // Initialization of the char array with 6 positions

    WiFiClient spClient // Create the object
    PubSubClient client (espClient) // Instantiate the MQTT client by passing the espClient object

    //// Sample query
    char INSERT_SQL [] = "INSERT INTO MQTT.TESTES (Temperature) VALUES (% .2f)";
    char query [255];

    MySQL_Connection conn (& espClient);
    MySQL_Cursor * cursor;

    void setup () {
      Serial.begin (115200);
      referenceVoltage = 1; // Reference voltage of ESP82666

      / * Connection to the Wi-Fi network * /
      Serial.printf ("\ nConnecting to% s", ssid);
      WiFi.begin (ssid, password);
      while (WiFi.status ()! = WL_CONNECTED) {
        delay (500);
        Serial.print (".");
      }
      / * Connection to MQTT * /
      client.setServer (mqttServer, mqttPort);
      while (! client.connected ()) {
        Serial.println ("Connecting to MQTT ...");
        if (client.connect ("ESP8266Client", mqttUser, mqttPassword)) {
          Serial.println ("connected to the MQTT");
        } else {
          Serial.print ("failed with state");
          Serial.print (client.state ());
          delay (5000);
        }
      }

      // print out info about the connection:
      Serial.println ("\ nConnected to network");
      Serial.print ("My IP address is:");
      Serial.println (WiFi.localIP ());
      / * Connection to the SQL database * /
      Serial.println ("Connecting to SQL ...");
      if (conn.connect (server_addr, 3306, login_username, login_password))
        Serial.println ("OK.");
      else
        Serial.println ("FAILED.");

      // create the MySQL object cursor
      cursor = new MySQL_Cursor (& conn);

    }

void loop () {
      / * Sending data to MQTT * /
      reading = 0;
      for (int i = 0; i <10; i ++) {// Average of 10 readings to make the reading more accurate
        reading + = analogRead (tempPin);
        delay (20); // A0 pin reading time every 20 milliseconds
      }
      // Conversion of analog voltage to temperature value in degrees Celsius and to an array
      tempC = (referenceVoltage * reading * 10) / 1023;
      convert + = tempC; // Converts the value of the variable tempC to string
      convert.toCharArray (array, 6) // convert the string into a char array with 6 positions

      client.publish ("esp8266", array);
      Serial.println (array);
      // convert = "";
      client.loop ();
      delay (2000);

      / * Sending data to the database * /
      read_value = 0;
      for (int i = 0; i <10; i ++) {// Average of 10 readings to make the reading more accurate
        read_value + = analogRead (tempPin);
        delay (20); // A0 pin reading time every 20 milliseconds
      }
      // Conversion of analog voltage to temperature value in degrees Celsius and to an array
      temp_C = (referenceVoltage * read_value * 10) / 1023;
      if (conn.connected ());
      sprintf (query, INSERT_SQL, temp_C);
      Serial.println (query);
      cursor-> execute (query);
      delay (2000);
    }
```
The serial monitor output is:

```
Connecting to network STARNAUTE4.
Connecting to MQTT...
connected to the MQTT

Connected to network
My IP address is: 192.168.43.18
Connecting to SQL...  
OK.
28.23
INSERT INTO MQTT.TESTES (Temperature) VALUES (28.30)
28.23
INSERT INTO MQTT.TESTES (Temperature) VALUES (28.25)
28.23
INSERT INTO MQTT.TESTES (Temperature) VALUES (28.18)
28.23
INSERT INTO MQTT.TESTES (Temperature) VALUES (28.24)
```
But not send any value to the broker MQTT and to the database mysql

smalhao, 2018-10-04:
``

devyte, 2018-10-05:
@smalhao 
What is this:
```
if (conn.connected ()); //<--  ; at the end
```
and why do you have the same calculation duplicate for the MQTT and for the db (averaging loop and convert to string)?

I don't see anything that jumps out at me. In any case, this is not a general support forum to get help with your particular project. This is an issue tracker, meant to track issues in the core hosted in this repo.  Also, if you have a problem with a 3rd party lib, I suggest requesting help from the lib's author.

Closing, see issue POLICY doc and issue template instructions for more details.

closed 2018-10-05 by devyte