esp8266/Arduino #5212: Add destructor to fs::File which will close the file

TD-er, 2018-10-06:
[Fs.h](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h) doesn't implement a destructor.
It would be nice I guess to have a destructor which calls the `close()` function, since it is not useful to keep the file open if the File object gets out of scope.
A forgotten call to `close()` seem to introduce at least a memory leak somewhere, but maybe also other issues I cannot think of yet.

So is it a good idea to add a (virtual) destructor for fs::File ?

Makuna, 2018-10-06:
The way its implemented, it relies on the smart pointer to the implementation to do any cleanup as it is needed.  It doesn't look like any of the implementations do anything in their destructors though.

earlephilhower, 2019-01-23:
We're redoing the FSImpl interface and this can slide in w/the other changes.  I don't think leaking in SPIFFS is possible (you have to pass in memory for the FD and for the buffer from the wrapper class), but it's simple enough to verify.

earlephilhower, 2019-04-15:
As far as I can see SPIFFs already calls `close()` on itself when deleted.
````
    ~SPIFFSFileImpl() override
    {
        close();
    }
````

Closing for now, but if you have a MCVE we can run and see something anomalous, please do post it and re-open.

closed 2019-04-15 by earlephilhower