esp8266/Arduino #5223: Issue with GPIO16

zeroprime790, 2018-10-11:
Hello,
I'm facing a really strange behavior with GPIO16

In the code bellow I attach the Tx of SoftwareSerial to GPIO 16 (D0) and it works just fine. But when I setup pins D5,D7,D6 the SoftwareSerial stop working.

  // set the data rate for the SoftwareSerial port
  mySerial.begin(9600);
  delay(5000);
  mySerial.write("b");
  //button
  pinMode(buttonPin, INPUT);
  //RGB
pinMode(greenLed, OUTPUT);
pinMode(redLed, OUTPUT);
pinMode(blueLed, OUTPUT);

 If I remove these three pin modes, Pin 16 works despite the interrupts. No matter where I set the pins (setup, loop...) the softwareSerial communication will stop working. I even tried to begin the serial inside the loop without any luck.

`#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include "SoftwareSerial.h"
#include <Wire.h>

#define PWM_OUT  D2  // Fan speed control
#define FADE_DELAY 1000
#define redLed D5
#define blueLed D7
#define greenLed D6

//ATTINY85 COMMUNICATION
SoftwareSerial mySerial(-1, D0); // RX, TX

//FAN RPM
<...>
int hallsensor = D3;
<...>
//BUTTON
int buttonPin = D4;
<...>
const char *ssid = "***";
const char *password = "***";
<...>
ESP8266WebServer server ( 80 );

String getPage(){
 <...>
}

<...>
void setup ( void ) {

  Serial.begin ( 115200 );
  WiFi.begin ( ssid, password );
  Serial.println ( "" );

  // Wait for connection
  while ( WiFi.status() != WL_CONNECTED ) {
    delay ( 500 );
    Serial.print ( "." );
  }

  Serial.println ( "" );
  Serial.print ( "Connected to " );
  Serial.println ( ssid );
  Serial.print ( "IP address: " );
  Serial.println ( WiFi.localIP() );
  
  pinMode(hallsensor, INPUT);

  attachInterrupt(D3, pickrpm, RISING);

  // Set Fan speed
  analogWrite( PWM_OUT, 360);
  
  // set the data rate for the SoftwareSerial port
  mySerial.begin(9600);
  delay(5000);
  mySerial.write("b");
  //button
  pinMode(buttonPin, INPUT);
  //RGB
pinMode(greenLed, OUTPUT);
pinMode(redLed, OUTPUT);
pinMode(blueLed, OUTPUT);

	server.on ( "/", handleRoot );

	server.begin();
	Serial.println ( "HTTP server started" );
}

void loop ( void ) {

  server.handleClient();
  //FAN RPM
  ticks = 0;      // Make ticks zero before starting interrupts.
  interrupts();    // or use sei(); to Enables interrupts
  delay (1000);     //Wait 1 second
  noInterrupts();  //  or use  cli();  to Disable interrupts
  Speed = ((ticks * 60) / fanspace[fan].fandiv);
  //Print calculated Speed at the serial port
  Serial.print (Speed, DEC);
  Serial.print (" RPM\r\n");
  // Converting integer Speed to an ASCII string of 4 characters padded left
  char  SpeedString[4];    // Buffer to store string of 4 chars + 0 termination
  sprintf(SpeedString, "%4d", Speed);   // change this to %3, 4 ,5 etc depending upon your max speed
  s = Speed;

  delay(1000);
}`

5chufti, 2018-10-11:
where is the fileld-in template that you were offered when opening the issue?
amongst others things you were asked to present a minimum code example  to evaluate your problem.
Your presented code contains things that definitely are much above the minimum needed.
cut it down to the bare minimum an see if the problem persists, than report back correctly with filled in template.
So for now, please close the issue

zeroprime790, 2018-10-11:
ok, I will try the minimum code. Thank you

closed 2018-10-11 by zeroprime790