esp8266/Arduino #5229: GPIO16 issue with SoftwareSerial

zeroprime790, 2018-10-11:
- Hardware:				ESP12-E
- Core Version:			2.4.2
- Development Env:		Arduino IDE 1.8.7

### Settings in IDE

- Module:  			NodeMCU 1.0
- Flash Size: 		        4M
- lwip Variant:                  v2 Lower Memory
- CPU Frequency:		80Mhz

### Problem Description

Hello,
I'm facing a really strange behavior with GPIO16

In the code bellow I attach the Tx of SoftwareSerial to GPIO 16 (D0) and it works just fine. But when I setup pins D5 and D7 the SoftwareSerial stop working.

If I don't set pins D5 and D7, Pin 16 and every other pin works. 
No matter where I set the pins (setup, loop...) the softwareSerial communication will stop working.
I even tried to begin the serial inside the loop without any luck.

And something else that is driving me mad:
**If I set for example D5 or D7 pins as INPUT the SoftwareSerials works just fine**

Also, I tries to load the espsoftwareserial library but the result remains the same
https://github.com/plerup/espsoftwareserial

### Sketch

```cpp

#include <SoftwareSerial.h>
#include <Wire.h>

#define redLed D5
#define blueLed D7
#define greenLed D6

//ATTINY85 COMMUNICATION
SoftwareSerial mySerial(-1, D0); // RX, TX

void setup ( void ) {
  
  // set the data rate for the SoftwareSerial port
  mySerial.begin(9600);
  delay(5000);
  mySerial.write("b");
  pinMode(greenLed, OUTPUT);
  pinMode(blueLed, OUTPUT);
  pinMode(redLed, OUTPUT);

}

void loop ( void ) {
  
  mySerial.write("r");
  delay(5000);
  mySerial.write("g");
  delay(5000);
  mySerial.write("w");  
}

### Debug Messages

The Sketch compiles without any errors

devyte, 2018-10-12:
Our SoftwareSerial comes from https://github.com/plerup/espsoftwareserial .
What happens at a higher baud, e.g.: 115kbps?

devyte, 2018-10-12:
@plerup do you have any insight here? I don't have any way of testing at my end.

5chufti, 2018-10-12:
looking at the code, it obviously is not prepared to use gpio16 as it can't generate interrupts.
Why it displays this interesting behaviour would need deeper analysis.
But this is more likely to be handled as [issue over there](https://github.com/plerup/espsoftwareserial/issues) ...

plerup, 2018-10-13:
Sorry, but I have no idea and I don't have a NodeMCU card.
However GPIO16 is special as it is not possible to connect an interrupt to it. That's why espsoftwareserial only allows it to be Tx. Don't know if there are any other side effects.

earlephilhower, 2019-07-25:
This is a SWSerial issue, not a core one, and it is due to the physical hardware of the ESP8266.  GPIO16 is not part of the core, but part of the RTC unit and as the SWSerial author said, it's "special" (as in slower than other GPIOs, cannot have IRQ associated with it, etc.).  Closing.

closed 2019-07-25 by earlephilhower