esp8266/Arduino #5231: Cannot change station MAC

num19, 2018-10-11:
It

`extern "C" {
  #include "user_interface.h"
  #include "wpa2_enterprise.h"
}
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

// SSID to connect to
static const char* ssid = "SSID";
// Username for authentification
static const char* username = "USERNAME";
// Password for authentification
static const char* password = "PASSWORD";
const int ledPin = 0;
// --------------------------------------------------------------

void setup() {
  uint8_t mac[6] {0x1A, 0x1A, 0x1A, 0x1A, 0x1A, 0x1A};
  //WiFi.macAddress(mac);
  wifi_set_macaddr(STATION_IF, mac);
  Serial.begin(115200);
  Serial.println();
  Serial.println(WiFi.macAddress()); // prints old mac
  pinMode(ledPin, OUTPUT);

  Serial.print("Tryingonnect to ");
  Serial.println(ssid);

  wifi_station_disconnect();
  struct station_config wifi_config;

  memset(&wifi_config, 0, sizeof(wifi_config));
  strcpy((char*)wifi_config.ssid, ssid);
  strcpy((char*)wifi_config.password, password);
  wifi_station_set_config(&wifi_config);
  wifi_set_macaddr(STATION_IF, mac);
  Serial.println(WiFi.macAddress()); // prints also old mac

  wifi_station_set_wpa2_enterprise_auth(1);
  wifi_station_set_enterprise_username((uint8*)username, strlen(username));
  wifi_station_set_enterprise_password((uint8*)password, strlen(password));
  wifi_station_connect();

  Serial.print("Status: ");
  Serial.println(wifi_station_get_connect_status());

  // Wait for connection AND IP address from DHCP
  while (WiFi.status() != WL_CONNECTED) {
    delay(2000);
    Serial.println(WiFi.localIP());
  }

} // setup

// ---------------------------
void loop() {
 
}`

dragondaud, 2018-10-12:
This was answered previously, see [Issue#413](https://github.com/esp8266/Arduino/issues/413#issuecomment-113941647).
I notice in the working example they set Station mode, explicitly.
You could also check the return status of wifi_set_macaddr() to see if it succeeded.

devyte, 2018-10-12:
The answer in #413 is that
1. You have to change it before calling WiFi.begin()
2. The first byte of the mac must be even

Both conditions seem to be met in the above sketch.
However, disconnect() is called after setting the mac, which makes me think of persist, I.e.: wifi could have started before setup() with parameters stored in the sdk. I would say call disconnect() first, then change the mac, then start the WiFi.
Having said that, the code above is using sdk api and not our core. Therefore, if there is an issue with the functionality, it's not in our core code.
Closing.

closed 2018-10-12 by devyte