esp8266/Arduino #5233: Complete lockup of ESP8266 with 2.4.2

AcuarioCat, 2018-10-12:
- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP12
- Core Version:			2.4.2
- Development Env:		Visual Micro
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266
- Flash Mode:			qio
- Flash Size: 		        4MB
- lwip Variant:                 v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	      40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:            115200

### Problem Description
I am experiencing a total lockup of the ESP8266 when using core version 2.4.2. The chip seems to get to a state where no watchdog timer is fired, neither my own (using the Ticker class), the esp internal software or hardware watchdogs. There is no serial output, nothing. The only way to recover is a hard reset. I have serial debugging enabled (Serial.setDebugOutput(true);) but this shows nothing.  In my standard debug output I report the free heap and this is showing over 26K free.  The worrying thing is that the esp watchdogs aren't firing.

I have tested this on 2 different devices (to eliminate hardware failure) and the same problem occurs. The time the device is running is variable, it can be a few minutes or a few hours.

The identical code compiled using core 2.4.0 does not have this problem. 

The devices are powered from a stabilized bench psu so it is unlikely to be a power problem. 

I'm at a loss as to what more information I can give or what more I can do to try to track down the problem. Any suggestions welcome.

devyte, 2018-10-12:
You could start withan MCVE sketch as instructed in the issue template. Also, try to eliminate all 3rd party libs, and leave only core code. If you find the issue is caused by a 3rd party lib, then please report it there first.

AcuarioCat, 2018-10-13:
I will try to create a MCVE sketch but it's a very big project with lots of modules.
I have now tried with the core 2.4.1 and this also exhibits the same problem.
Again, this does not occur in 2.4.0 and there is no modification done to any of the 3rd party libs to get it to run on the later builds.

I found reference to a sleep problem with Sonoff-Tasmota that seems similar. Checking the power consumption I see a big drop when the ESP stops responding. Testing with code to force it to deep sleep it would appear this is the problem however when I force a sleep in code I do get debug output:
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
enter deep sleep

When it seems to go to sleep without asking I get no debug output.
I have tried adding WiFi.setSleepMode(WIFI_NONE_SLEEP); but that makes no difference.

Sometimes it runs for hours before 'sleeping', other times it happens in a few minutes.

devyte, 2018-10-13:
Try with latest git. The sdk has a known issue with light sleep post 2.4.0, and it was addressed only recently in the sdk. Then, the sdk was updated here, with related code changes on our side, only a few days ago.

devyte, 2018-10-13:
Release 2.4.2 has an optimization enabled that causes corruption when using wps. This was fixed on in latesy git, where the optimization is automagically disabled.
Releases 2.4.0 and 2.4.1 have other stability issues.
I suggest sticking.with latest git.

AcuarioCat, 2018-10-13:
Do you mean the release or download of the current source? The above WiFi problem is with the current downloaded source (Arduino-master). Reverting back to a release (2.4.2) and the problem goes away.

d-a-v, 2018-10-13:
The SDK has changed since 2.4.2. 
we were using nonos-sdk 2.1 
we have now sdk-pre3.0
Do you use WEP ?

AcuarioCat, 2018-10-13:
Yes I do but I can easily remove it to try (and remove the extra 4K stack usage). The WEP code is only run if the esp fails to connect to the AP. I can try it with hard coded ssid and key and report back.

d-a-v, 2018-10-14:
You are speaking of WPS (extra 4K is available when WPS is *not* used).
Anyway trying with hardcoded ssid/psk is welcome,

I'm speaking of WEP. In nonos-sdk v3, they disabled WEP by default (WiFi security protocol - weak).
To use WEP as client, ts should be explicitly enabled with esp api, this is not documented yet.

AcuarioCat, 2018-10-16:
I'm now using the latest git and the sleep problem seems to have stopped happening. I'll close the issue for now.

closed 2018-10-16 by AcuarioCat

arihantdaga, 2018-11-12:
@AcuarioCat How did you fix this..

AcuarioCat, 2018-11-12:
I never really found what was causing the problem. In the end downloading and using the latest GIT worked. Whether something was changed somewhere that fixed the problem I can not tell. Are you suffering the same problem?

arihantdaga, 2018-11-12:
I am facing frequent hardware watchdog reset problems. (I think its something to do with the serial). Later i added this in my code after doing serial.println() `optimistic_yield(100);` That seems to have fixed this problem...

Spackstor, 2019-01-09:
I have the same or similar to AcuarioCat (no watchdog issues) using the ESP8266-12e. I ran into a solution that uses a timeout (infinite loop prevention?) for the while(!client..., This appears to fix lockup, but it prevents "Marc Speybrouck's timer" from working. In another program the timeout impacts IR counts on A0 for a locator wheel. So my quest for a solution continues.
Following this thread I went from 2.4.0 to 2.4.2. This appears to prevent lockup but impacts the timer in the same way. 
(Using the ESP8266 just for WiFi doing TxRx to a nano, fixes the problem but using a single processor is the objective.)
I post this because for my purposes 2.4.2 appears to embed a timeout, impacting control of other events. 
In short my programs work with 2.4.0  but freeze anywhere from minutes to as much as 10 days. With 2.4.2 they so far do no freeze but time consuming objects controlling items outside the WiFi objects  (the purpose of the programs) no longer work properly. So I may get 2.4.0 to work but 2.4.2 simply does not.