esp8266/Arduino #5236: error compiling digest example

notsolowki, 2018-10-13:
when i try to compile the digest authentication exaple im getting error

exit status 1
cannot declare variable 'client' to be of abstract type 'WiFiClient'

dragondaud, 2018-10-13:
You did not use issue template, but with current git version of core the DigestAuthorization example compiles fine. Please close, or provide details.

notsolowki, 2018-10-13:
I'm using the git version of the core on 2 different computers . Just to
make sure I installed arduino ide on another computer and get the same
result.

notsolowki, 2018-10-13:
sorry im not sure how to use the issue template maybe this will help?

`
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\User\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\User\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,VTable=flash,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=512K0,led=2,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10803 -build-path C:\Users\User\AppData\Local\Temp\arduino_build_525688 -warnings=none -build-cache C:\Users\User\AppData\Local\Temp\arduino_cache_534053 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -verbose C:\Users\User\Documents\Arduino\esp8266_digest_client\esp8266_digest_client.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\User\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\User\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,VTable=flash,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=qio,FlashSize=512K0,led=2,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10803 -build-path C:\Users\User\AppData\Local\Temp\arduino_build_525688 -warnings=none -build-cache C:\Users\User\AppData\Local\Temp\arduino_cache_534053 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -verbose C:\Users\User\Documents\Arduino\esp8266_digest_client\esp8266_digest_client.ino
Using board 'generic' from platform in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Using core 'esp8266' from platform in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2
Detecting libraries used...
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\User\AppData\Local\Temp\arduino_build_525688/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10803 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\generic" "C:\Users\User\AppData\Local\Temp\arduino_build_525688\sketch\esp8266_digest_client.ino.cpp" -o "nul"
"C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lwip2/include" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\User\AppData\Local\Temp\arduino_build_525688/core" -c -w  -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10803 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266" "-IC:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\variants\generic" "-IC:\Users\User\Documents\Arduino\libraries\ESP8266WiFi\src" "C:\Users\User\AppData\Local\Temp\arduino_build_525688\sketch\esp8266_digest_client.ino.cpp"  [... cut ...]
 Not used: C:\Users\User\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\SPI
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\User\Documents\Arduino\libraries\ESP8266WiFi 
Using library ESP8266HTTPClient at version 1.2 in folder: C:\Users\User\Documents\Arduino\libraries\ESP8266HTTPClient 
Using library SPI at version 1.0 in folder: C:\Users\User\Documents\Arduino\libraries\SPI 
exit status 1
cannot declare variable 'client' to be of abstract type 'WiFiClient'
`

notsolowki, 2018-10-13:
the code tags didnt work very well. ive also used the versions of the libraries in appdata

notsolowki, 2018-10-13:
im not the only person getting an error https://forum.arduino.cc/index.php?topic=573306.0

devyte, 2018-10-13:
It seems to me you're trying to use the digest example downloaded from the repo webpage with the core you have installed from the board package. Is that correct?

notsolowki, 2018-10-13:
i used board manager to install the esp8266 and then copied everything from the library folder here into my library folder in my documents folder

devyte, 2018-10-13:
Oh boy. You're trying to use latest git examples with your 2.4.2 release version of the core. Whatever possessed you to try that?
The core is released with the examples. When something in the core is updated, the examples are updated as well if needed. You ALWAYS use the examples with the core version they belong to.
If you want to use the examples shown in in the latest git here, you have to install latest git of the core, not the release.
Closing due to user error.

closed 2018-10-13 by devyte

devyte, 2018-10-13:
No, you're not, your logs clearly show building from 2.4.2. Did you uninstall that before building git?