esp8266/Arduino #5238: DigestAuthorization.ino does not compile

notsolowki, 2018-10-13:
`C:\Users\User\Documents\Arduino\esp8266_digest_client\esp8266_digest_client.ino: In function 'void loop()':

esp8266_digest_client:100: error: no matching function for call to 'HTTPClient::begin(WiFiClient&, StringSumHelper&)'

   http.begin(client, String(server) + String(uri));

                                                  ^

C:\Users\User\Documents\Arduino\esp8266_digest_client\esp8266_digest_client.ino:100:50: note: candidates are:

In file included from C:\Users\User\Documents\Arduino\esp8266_digest_client\esp8266_digest_client.ino:9:0:

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:137:10: note: bool HTTPClient::begin(String)

     bool begin(String url);

          ^

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:137:10: note:   candidate expects 1 argument, 2 provided

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:138:10: note: bool HTTPClient::begin(String, uint16_t, String)

     bool begin(String host, uint16_t port, String uri = "/");

          ^

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:138:10: note:   no known conversion for argument 1 from 'WiFiClient' to 'String'

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:140:10: note: bool HTTPClient::begin(String, String)

     bool begin(String url, String httpsFingerprint);

          ^

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:140:10: note:   no known conversion for argument 1 from 'WiFiClient' to 'String'

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:141:10: note: bool HTTPClient::begin(String, uint16_t, String, String)

     bool begin(String host, uint16_t port, String uri, String httpsFingerprint);

          ^

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:141:10: note:   candidate expects 4 arguments, 2 provided

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:143:10: note: bool HTTPClient::begin(String, const uint8_t*)

     bool begin(String url, const uint8_t httpsFingerprint[20]);

          ^

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:143:10: note:   no known conversion for argument 1 from 'WiFiClient' to 'String'

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:144:10: note: bool HTTPClient::begin(String, uint16_t, String, const uint8_t*)

     bool begin(String host, uint16_t port, String uri, const uint8_t httpsFingerprint[20]);

          ^

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:144:10: note:   candidate expects 4 arguments, 2 provided

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:146:10: note: bool HTTPClient::begin(String, uint16_t, String, bool, String)

     bool begin(String host, uint16_t port, String uri, bool https, String httpsFingerprint)  __attribute__ ((deprecated));

          ^

C:\Users\user2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.h:146:10: note:   candidate expects 5 arguments, 2 provided

exit status 1
no matching function for call to 'HTTPClient::begin(WiFiClient&, StringSumHelper&)'
`
not sure while this wont compile

notsolowki, 2018-10-13:
and most of the example in httpclient don't compile either

devyte, 2018-10-13:
I have already explained: you have 2.4.2 release installed (shown by your logs), and you're trying to compile the examples downloaded from this webpage, which are NOT for 2.4.2, but for latest git.
Delete everything, install git cleanly, and DON'T download any code manually.
Closing.

closed 2018-10-13 by devyte

notsolowki, 2018-10-13:
okay im sorry i was unaware of the releases differences. can i use git on windows?

notsolowki, 2018-10-13:
can you please give me some instruction to install the git on windows??

notsolowki, 2018-10-13:
here is swhat happens when i try to install binary tools 
C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools>python get.py
Platform: i686-mingw32
Downloading esptool-0.4.13-win32.zip
Traceback (most recent call last):
  File "get.py", line 116, in <module>
    get_tool(tool)
  File "get.py", line 75, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\urllib.py", line 91, in urlretrieve
    return _urlopener.retrieve(url, filename, reporthook, data)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\urllib.py", line 237, in retrieve
    fp = self.open(url, data)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\urllib.py", line 205, in open
    return getattr(self, name)(url)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\urllib.py", line 435, in open_https
    h.endheaders(data)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\httplib.py", line 940, in endheaders
    self._send_output(message_body)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\httplib.py", line 803, in _send_output
    self.send(msg)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\httplib.py", line 755, in send
    self.connect()
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\httplib.py", line 1156, in connect
    self.sock = ssl.wrap_socket(sock, self.key_file, self.cert_file)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\ssl.py", line 342, in wrap_socket
    ciphers=ciphers)
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\ssl.py", line 121, in __init__
    self.do_handshake()
  File "C:\Users\User\AppData\Local\Arduino15\packages\esp8266\tools\lib\ssl.py", line 281, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [Errno 1] _ssl.c:499: error:1407742E:SSL routines:SSL23_GET_SERVER_HELLO:tlsv1 alert protocol version

notsolowki, 2018-10-13:
when i get clone your code i still get error 

exit status 1
Error compiling for board Generic ESP8266 Module.