esp8266/Arduino #5239: Wifi Client is not connected but wifi.status() ==3

arnold-b, 2018-10-13:
Hello, 

the Situation is i try to connect to wifi SSID wodewanluo. 
After a few tries. ESP said suddenly its connected, even the SSID don´t exist. 
wifiState ist a state counter to change between different wifi config´s

TimerEvent::wifiState 1 Wifi.status 3
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 5
Auto connect: 1
SSID (10): wodewanluo
Passphrase (10): xxxx
BSSID set: 0

TimerEvent::Ticker 335
TimerEvent::Ticker 336
TimerEvent::Ticker 337
TimerEvent::Ticker 338
TimerEvent::Ticker 339
TimerEvent::wifiState 1 Wifi.status 3
Mode: STA
PHY mode: N
Channel: 11
AP id: 0
Status: 5
Auto connect: 1

arnold-b, 2018-10-13:
Serial.println(WiFi.getMode()) = 1 => WIFI_STA
	Serial.println(WiFi.isConnected()); =0
	Serial.println(wifi_station_get_connect_status()); = 1
	Serial.println(wifi_station_disconnect()); 1

Finaly its now in AP mode  because he created the Wifi AP and i can access it. 
But the Mode shows STA
isConnected shows falls but WiFi.status shows 3 (connected).

d-a-v, 2021-03-31:
Closing as duplicate #7432, let's follow-up there !

closed 2021-03-31 by d-a-v