esp8266/Arduino #5242: WDT Softreset  when Uploading File to SPIFFS

Ikswoknib, 2018-10-13:
### Basic Infos

- [ X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X ] I have tested that the issue is present in current master branch (aka latest git).
- [ X ] I have searched the issue tracker for a similar issue.
- [ X ] If there is a stack dump, I have decoded it.
- [ X ] I have filled out all fields below.

#### Platform

- Hardware:			Wemos D1 mini
- Core Version:			Releas V2.4.2
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Lolin Wemos D1 mini
- Flash Size: 		4MB(3MB SPIFFS)
- lwip Variant:   v2 Lower Memory
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I have a Sketch where I upload a File through an simple HTML Page, but while or after doing that the Program makes an WDT Software reset. This is only at Files above 100kb.
I used the ESPNexUpload Demo first. With this one it worked without a problem. I began to modify it for my purposes, but i haven't changed the Upload Part. But now i get this error.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Only Main Part and the needed parts for uploading:

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPClient.h>

#include <FS.h>
#include <SoftwareSerial.h>
#include <ESPNexUpload.h>
#include <ArduinoOTA.h>

#include <WiFiUdp.h>
SoftwareSerial softSerial(5, 4); /* For Wemos D1 mini RX:D1/5, TX:D2/4 */

String req;
String ssid = "";
String password = "";
String host = "";
String url = "";
String timetosleep = "";
String wakeontouch = "";
String nexwifistatus = "";

bool WifiConnected = false;

bool ConnectionEstablished; // Flag for successfully handled connection
char buffer[200];  

#define MAX_TELNET_CLIENTS 2
 
WiFiServer TelnetServer(22);
WiFiClient TelnetClient[MAX_TELNET_CLIENTS];

#define DEBUG_ESP_OTA;

[... 307 lines cut ...]
0x40208c93: handleFileUpload() at D:\Windows_Dokumente\Arduino\ESP2Nextionv3/FileServer.ino line 63
0x401006dc: free at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1755
0x4010020c: _umm_free at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1295
0x4021431e: std::_Function_handler ::_M_invoke(std::_Any_data const&) at c:\users\christian\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x4020c03e: std::function ::operator()() const at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ArduinoOTA/ArduinoOTA.cpp line 68
0x4020b350: FunctionRequestHandler::canUpload(String) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 221
0x4020c0b9: FunctionRequestHandler::upload(ESP8266WebServer&, String, HTTPUpload&) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 221
0x40212678: String::String(String const&) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 566
0x4020ce9f: ESP8266WebServer::_parseForm(WiFiClient&, String, unsigned int) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/Parsing.cpp line 514
0x4020cd70: ESP8266WebServer::_parseForm(WiFiClient&, String, unsigned int) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/Parsing.cpp line 495
0x402124cb: String::changeBuffer(unsigned int) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 566
0x4020c08c: FunctionRequestHandler::upload(ESP8266WebServer&, String, HTTPUpload&) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 221
0x40212650: String::operator=(String const&) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 566
0x4020d829: ESP8266WebServer::_parseRequest(WiFiClient&) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/Parsing.cpp line 218
0x401066d0: millis at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_wiring.c line 183
0x4020b458: FunctionRequestHandler::canHandle(HTTPMethod, String) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 221
0x4020c33b: ESP8266WebServer::handleClient() at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer\src/ESP8266WebServer.cpp line 221
0x40212596: String::String(char const*) at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 566
0x4020a090: ESP8266WiFiSTAClass::status() at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.cpp line 381
0x402124a4: String::~String() at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/WString.cpp line 566
0x402095c0: loop at D:\Windows_Dokumente\Arduino\ESP2Nextionv3/ESP2Nextionv3.ino line 133
0x402089e1: setup at D:\Windows_Dokumente\Arduino\ESP2Nextionv3/ESP2Nextionv3.ino line 126
0x402132ac: loop_wrapper at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 125
0x40100739: cont_wrapper at C:\Users\Christian\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/cont.S line 81
```

TD-er, 2018-10-17:
I do also get a lot of hardware watchdog reboots and I really don't have a clue what is happening in my situation. So I am very interested in what is causing this and this is my way to 'follow' this issue.

JohannHaller, 2018-11-02:
Let me guess: the ESP8266 Version you have installed in Arduino is >=2.4
Try 2.3...
[image]

TD-er, 2018-11-02:
@JohannHaller Does ESPeasy still compile with core 2.3.0?

d-a-v, 2018-11-02:
With current git, !m!c!vE above made into MCVE below, with 3MB-SPIFFS and 2MB file,

```
$ curl -F data=@/home/david/photo-id.eps http://192.168.1.174
                                          Connected! IP address: 192.168.1.174
                                          Starting Http server
                                          Starting OTA server
                                          handleFileUpload Name: /photo-id.eps
                                          handleFileUpload Size: 1890751
                                          Restarting ESP
$ cd /tmp
$ wget http://192.168.1.174/photo-id.eps
                                          Connected! IP address: 192.168.1.174
                                          Starting Http server
                                          Starting OTA server
                                          handleFileRead: /photo-id.eps
                                          	Sent file: /photo-id.eps
$ md5sum photo-id.eps ~/photo-id.eps
a72f6e309c4d3b931b0cb5f89424869e  photo-id.eps
a72f6e309c4d3b931b0cb5f89424869e  /home/david/photo-id.eps
```

@Ikswoknib There is indeed an issue with 2.4.2, but it seems to be fixed in git version. Please try and report.
@JohannHaller I hope incoming v2.5.0 will fulfill your requirements.

Compiling sketch:
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPClient.h>

#include <FS.h>
#include <SoftwareSerial.h>
//#include <ESPNexUpload.h>
#include <ArduinoOTA.h>

#include <WiFiUdp.h>
SoftwareSerial softSerial(5, 4); /* For Wemos D1 mini RX:D1/5, TX:D2/4 */

String req;
String ssid = "SSID";
String password = "PSK";
String host = "test5242";
String url = "test";
String timetosleep = "tts";
String wakeontouch = "wot";
String nexwifistatus = "nws";

bool WifiConnected = false;

bool ConnectionEstablished; // Flag for successfully handled connection
char buffer[200];  

#define MAX_TELNET_CLIENTS 2
 
WiFiServer TelnetServer(22);
WiFiClient TelnetClient[MAX_TELNET_CLIENTS];

//#define DEBUG_ESP_OTA;

HTTPClient http;

char inChar;
// Name for updatefile (no need to change, used only internally)
String updateFileName = "/update.tft";

ESP8266WebServer server(80);

[... 186 lines cut ...]
    if(fsUploadFile)
      fsUploadFile.write(upload.buf, upload.currentSize); // Write the received bytes to the file

  } else if(upload.status == UPLOAD_FILE_END){
    if(fsUploadFile) {                                    // If the file was successfully created
      fsUploadFile.close();                               // Close the file again
      Serial.print("handleFileUpload Size: "); Serial.println(upload.totalSize);
      //TelnetMsg("handleFileUpload Size: " + upload.totalSize);
      //Wenn .tft File dann UpdateNextion
      if(upload.filename.endsWith(".tft")){
        Serial.println("Sending file to display");
        //TelnetMsg("Sending file to display");
        //updateNextion();
      } else { 
        //TelnetMsg("Restarting ESP");        
        Serial.println("Restarting ESP");        
        ESP.restart();
      }
    } else {
      server.send(500, "text/plain", "500: couldn't create file");
      //TelnetMsg("500: couldn't create file");
    }
  }
}
```

earlephilhower, 2019-01-17:
@d-a-v, can we close this?  Looks like GIT head fixed it a while back.

d-a-v, 2019-01-17:
No report/answer for two months after a proper working MCVE. No news, good news.
I guess another issue could be opened if the issue persists.

closed 2019-01-17 by d-a-v