esp8266/Arduino #5246: SPIFFS crashes Fatal exception 0(IllegalInstructionCause) if SPIFFS not assigned

AcuarioCat, 2018-10-14:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:				WemosD1 and NodeMCU V1.0
- Core Version:			12/10/2018 00:34
- Development Env:		Visual Micro
- Operating System:		Windows

### Problem Description

More of a 'gotcha' than a problem so it turns out.
On previous builds if a SPIFFS region wasn't assigned calling SPIFFS.begin() worked silently (ok, I guess there would have been problems later on).
On the current git calling begin() causes:
Fatal exception 0(IllegalInstructionCause):
epc1=0x4000e25d, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

As I said, more of a gotcha than a bug but it would be kinder if there was some sort of compiler warning.

scropion86, 2018-10-14:
without s sketch mostly you cant get more help , try to completely erase your ESP Flash and re-upload your sketch if you still have an issue post your code here to look at it

AcuarioCat, 2018-10-14:
Here it is:
#include <ESP8266WiFi.h>
#include <FS.h>

void setup()
{
	Serial.begin(115200);
	delay(1000);
	
	Serial.println("Start SPIFFS");
	SPIFFS.begin();
}

void loop()
{
}

Set your build up to 4MB (no SPIFFS), build with the lates git and run it - continual crashes.
Do the same with 2.4.2 (or any previous build), no crash.

If you set the build to (for example) 4MB (1MB SPIFFS), no problem.

As I said, it's a gotcha but it would be good to have some sort of warning if this is possible.

d-a-v, 2018-10-14:
@scropion86 @AcuarioCat thanks for MCVE. bug reproduced.

PR #5249 (2018-12-06): fix SPIFFS when not enabled in build options
fix #5246 
~also  may help debugging #2581 
by adding optional `SPIFFS.begin(false)`: will not try to format (default is true)~
closed 2018-12-06 by d-a-v