esp8266/Arduino #5251: How to save WiFi credentials after smartconfig is done?

sabersingh, 2018-10-16:
I was using smartconfig feature in my project but I wonder how can I save the wifi SSID and Password that I receive from the app in SPIFFS. I mean it automatically connects ESP to wifi on receiving the UDP packets, If someone can tell me how can I get those credentials to store them in the file system or if I can parse those UDP packets somehow which are received by the ESP.

Thanks in Advance.

devyte, 2018-10-17:
This is not the right place to ask such questions. This is an issue tracker, meant to track issues in the code hosted in this repo. When you opened this issue you were presented with an issue template with specific instructions. The first line explains this.
Closing. See issue POLICY doc and issue template for more details.

closed 2018-10-17 by devyte

sabersingh, 2018-10-17:
I am so sorry, I am totally new to github. I will definitely keep it in mind from now onwards.

sonictl, 2022-06-03:
try this below
``` 
Serial.printf("SSID:%s\r\n", WiFi.SSID().c_str());
Serial.printf("PSW:%s\r\n", WiFi.psk().c_str());
```