esp8266/Arduino #5253: What's the best way to write table data on webpage from SPIFFS and update when client add or deletes a row

opolczer, 2018-10-17:
Hello all

I currently have a table on my index page on the ESP8266 and when a client connects I need to read from a file in SPIFFS and update the table with it's content.
I would also like to update it when the client adds or deletes a row.

I have managed to see how to append to a file in SPIFFS but what I can't figure out is how to get it's data out line by line and send this updated data to the client .. ?

What is the best way to go about this ?

Thank you

Opolczer

5chufti, 2018-10-17:
When you opened this issue you were presented with an issue template, in which the first line reads:
> If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is 
> related to hardware, please discuss at a community forum like esp8266.com.

As you didn't care to read/fill the template and your question is undoubtedly no issue with the core, please close this issue.

devyte, 2018-10-17:
Previous comment is correct, this is not the right place to ask such questions.
Closing. Please see issue POLICY doc and issue template instructions for more details.

closed 2018-10-17 by devyte