esp8266/Arduino #5257: Did WiFiClient.connected recently change for small payloads?

marcelstoer, 2018-10-17:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-WROOM-2
- Core Version:			2..4.2
- Development Env:		Arduino IDE
- Operating System:		macOS

### Settings in IDE

- Module:  			ESP-WROOM-2
- Flash Mode:			qio
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- Reset Method:		ck
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:             other

### Problem Description
This is a follow-up to https://github.com/ThingPulse/esp8266-weather-station/issues/140 which in turns follows up on as long thread in our support forum.

It appears that a (nested) HTTP/TCP parse loop that first checks for `client.connected()` and then for `client.available() > 0` **no longer** works reliably for small payloads.

What _seems_ to be happening on the networking stack is that "sometimes" the `WiFiClient` reports `connected() == false` immediately when the payload is sufficiently small. However, even though the client is no longer connected its internal buffer contains all the data and, hence, `client.available()` returns more than 0.

Can you confirm that such a double nested loop, which is a very common sketch template, should these days be changed to only check for `client.available() > 0`?

### Sketch fragment

```cpp
    // WiFiClient * client = http.getStreamPtr();
    // or something like
    // BearSSL::WiFiClientSecure client;
    // ...
    // if (client.connect(host, port))
    while(client.connected()) {
      while((size = client.available()) > 0) {
        // client->read() and do stuff
        // ...
        // give WiFi and TCP/IP libraries a chance to handle pending events
        yield();
      }
    }
  }
```

d-a-v, 2018-10-18:
It is true that semantic of `client.connected()` has changed some time ago.

Current semantic is:
- `client.connected() == true` when the tcp link is still opened, meaning that:
  - we can write on it (remote is listening)
  - new data may be later received
- `client.available() == true` means some data are still buffered and available for reading.

So `client.available()` can be true while `client.connected()` is not anymore, and that happens when remote sends data, then immediately closes before we read them all.

Thus a general inner communication loop *could* be:
```
    while (client.available() || client.connected())
    {
        if (client.available())
        {
            // client.available() bytes are immediately available for reading
            // warning: reading them *allows* peer to send more, so they should
            // be read *only* when they can immediately be processed, see below
            // for flow control
        }
        if (client.connected())
        {
            if (client.availableForWrite() >= N)
            {
                // TCP layer will be able to *bufferize* our N bytes,
                // and send them *asynchronously*, with by default
                // a small delay if those data are small
                // because Nagle is around (~200ms)
                // unless client.setNoDelay(true) was called.
                // 
                // In any case client.write() will *never* block here
            }
            else
            {
                // or we can send but it will take time because data are too
                // big to be asynchronously bufferized: TCP needs to receive
                // some ACKs to release its buffers.
                // That means that write() will block until it receives
                // authorization to send because we are not in a
                // multitasking environment

                // It is always OK to do this, client.availableForWrite() is
                // only needed when efficiency is a priority and when data
                // to send can wait where they currently are, especially
                // when they are in another tcp client.

                // Flow control:
                // It is also important to know that the ACKs we are sending
                // to remote are directly generated from client.read().
                // It means that:
                // Not immediately reading available data can be good for
                // flow control and avoid useless memory filling/overflow by
                // preventing peer from sending more data, and slow down
                // incoming bandwidth
                // (tcp is really a nice and efficient beast)
            }
        }
        
        // yield() is optional and harmless:
        // - available() calls it when there is no available data
        //   available() must be tested first when using "(available() || connected())"
        // - blocking write() calls it too
        yield();
    }
```

Historically, `connected()` was returning true even when tcp connection was just closed, and that caused some issues. I can find the exact origin of this change if that's desired.

@earlephilhower Is this this semantic the same when using BearSSL ?

We really should document that somewhere.
For core-2.5.0 we should revisit all examples and fix where relevant
(and also upgrade all axTLS examples to BearSSL).

Bodmer, 2018-10-19:
@d-a-v 
I noticed lost data at the end of a json message parsing so deduced that there was still data available after disconnection and [implemented the OR logic](https://github.com/Bodmer/DarkSkyWeather/blob/master/DarkSkyWeather.cpp#L98). However I do not understand why the calling order is important, ref you comment:

//   available() must be tested first when using "(available() || connected())"

Under what conditions is this test order important.

marcelstoer, 2018-10-20:
@d-a-v @Bodmer I propose we don't rely on such implementation details in our client implementation as they might change in the future w/o breaking API. Same goes for relying on how BearSSL and axTLS implemented `connected()`.

The documented template should be robust enough to work reliably regardless of such subtleties (hence, `yield()` being part of the blue print).

d-a-v, 2018-11-20:
Made it a PR #5355.
Are something or some details missing ?

PR #5355 (2018-11-20): document wificlient loop
fix #5257, rendered
- [there](https://github.com/d-a-v/Arduino/blob/loopdoc/doc/esp8266wifi/client-examples.rst#read-reply-from-the-server)
- and [there](https://github.com/d-a-v/Arduino/blob/loopdoc/doc/esp8266wifi/client-examples.rst#general-client-loop)
commit b504881b (2018-11-20): document wificlient loop (#5355)

fix #5257
closed 2018-11-20 by devyte