esp8266/Arduino #5258: WiFi, safe moment to start UDP

pi3dec, 2018-10-18:
From the WiFi event on WIFI_EVENT_STAMODE_GOT_IP: I am activating UDP and sending few messages over UDP. I get constant crashes when I write to UDP within the WiFiEvent routine. however if in my loop routine  I start UDP only after initialisation and note from the WIFI_EVENT_STAMODE_GOT_IP event everything works.

WiFi.mode(WIFI_STA);
WiFi.config(myIP, gateway, mask);
WiFi.onEvent(WiFiEvent);
WiFi.begin(ssid, password);
 while (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("Connection Failed! Rebooting...");
        restart();
    }

// start the UDP routine 

IS it a bug? Or the implementation feature and I should wait until the end of WiFievent interrupt to start UDP from the main code routine? Have a delay of few seconds or something else?
Thanks.

d-a-v, 2018-10-18:
Some logs are welcome with debug options enabled,
a reproducable MCVE,
the version of the core you are running,
well, all that is asked in the issue template that you have deleted :)

If you are looping beyond reasonable within your event handler, try adding `esp_yield()` in it.

d-a-v, 2018-10-20:
So you are sending data from within a callback.
But with a delay it works.
Can you try setting global `iCanSendNowWithoutDelay=true` in this callback, then in your main loop,
```
if (iCanSendNowWithoutDelay) {
        udp->beginPacket(serverIP, serverPort);
        udp->write(temp.c_str(),  temp.length());
        udp->endPacket();
        iCanSendNowWithoutDelay = false;
}

d-a-v, 2018-10-21:
Hard to tell without MCVE.

earlephilhower, 2019-01-17:
Three months without an MCVE, decoded stack dump, or update.  Nothing to do here.  Closing.

closed 2019-01-17 by earlephilhower