esp8266/Arduino #5264: Wish list :  Arduino ESP8266 support Espressif's ESP8266_NONOS_SDK release 3.0

TridentTD, 2018-10-19:
Espressif's ESP8266_NONOS_SDK release 3.0
now add partition table.

https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/documents/EN/%20Partition%20Table.md

Thank you.

devyte, 2018-10-25:
Work on this is already ongoing, but unfortunately updating to the new sdk doesn't seem to be straightforward as in previous sdk updates.
Translation: at this point it builds, but doesn't work, and it will take a while to get it operational, and a long while to stabilize.

brandond, 2018-12-29:
I see there are a couple existing issues open asking for an update to the latest version of the NONOS-SDK. Is this still on the radar?

https://github.com/espressif/ESP8266_NONOS_SDK/commits/release/v3.0.0

d-a-v, 2018-12-29:
check also #5521 
nonos-sdk-v3 "official" will happen after our release core-2.5.0.
Updating is not straightforward as it was with previous versions.
I also read some days ago that they might consider removing some restricting features about partition and size, but I can't find the link.