esp8266/Arduino #5271: MQTT disconnects in release 2.4.2 (works in 2.3.0)

ikin77, 2018-10-23:
Hi. I've a working project based on 2.3.0 core, which runs flawlessly. I'd like to upgrade to 2.4.2 core, but I can't because I'm having issues with MQTT connections that don't happend with 2.3.0 version.

Let me copy here some code:

============ CODE START ===================
`
/********************************************
   Clase que gestiona la comunicacion MQTT  *
 ********************************************/

#include <Timers.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include "time.h"
#include "MQTTManager.h"
#include "acqua-config.h"
extern "C" {
#include "user_interface.h"
}

WiFiClient mqttWifiClient;

MQTTManager::MQTTManager() {

    this->json = (char*)malloc(MQTT_MAX_PACKET_SIZE);
    this->bufferTopicID = (char*)malloc(40);
    this->mqttClientId = (char*)malloc(20);
    sprintf(this->mqttClientId,"Device%i", this->espID);    
    this->mqtt = new PubSubClient(MQTT_BROKER, MQTT_PORT,
                                                      this->onMessageUserCallback(),
                                                      mqttWifiClient);
      
}

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////  

void MQTTManager::run() {

  Serial.println("[MQTTManager] run() Start");
  if (WiFi.status() == WL_CONNECTED) {
       Serial.println("[MQTTManager] run() WIFI Ok");
       if(this->connect()) { // Solo mandamos algo si tenemos algún cliente
              Serial.println("[MQTTManager] run() Connected to MTTT broker");
              this->sendMessageStatusTopic();
       }      
      else {
              Serial.println("[MQTTManager] run() Not MQTT connection");
      }    
  }
  Serial.println("[MQTTManager] run() End");
}

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////  
// Called every main Arduino "loop"

void MQTTManager::loop() {
  yield();
  this->mqtt->loop();
}

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////  

boolean MQTTManager::sendMessageStatusTopic() {

    sprintf(this->json, "{some json - not important}");                
    sprintf(this->bufferTopicID, "dv/%i/status", this->espID);
    this->mqtt->publish(this->bufferTopicID, this->json, true);

}

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/***************************************************************
 ******* Funcion para conectar y reconectar con el broker MQTT
 ****************************************************************/

boolean MQTTManager::connect() {

      Serial.println("[MQTTManager] connect() Start");

      // Si no estaba conectado intentamos conectar    
      if(!this->mqtt->connected()) {

        Serial.println("[MQTTManager] connect() Wasn't connected");

        this->mqtt->disconnect();
        yield();
        // Se establece el topic "will".
        sprintf(this->bufferTopicID, "dv/%i/status", this->espID);
        if(this->mqtt->connect(this->mqttClientId, "<mqtt-user>", 
                                              "<mqtt-password>",this->bufferTopicID,0,true,"{\"md\":-1}")) {
          Serial.println("[MQTTManager] connect() Connection successful");
          this->onConnectUserCallback();
          return true;  
        }
        else {
          Serial.println("[MQTTManager] connect() Connection fail");
          return false;
        }        
      }
      else {
        Serial.println("[MQTTManager] connect() Already connected");
        return true;
      }  
}

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/***************************************************************
 ******* Constructor de la función callback que se ejecutará al establecer conexión MQTT
 ******* Se harán suscripciones a topics
 ****************************************************************/

void MQTTManager::onConnectUserCallback() {

      // Subscriptions
      sprintf(this->bufferTopicID, "dv/%i/+/request", this->espID);
      this->mqtt->subscribe(bufferTopicID, 0); 

      // Nos suscribimos a info de clientes Android activos
      sprintf(this->bufferTopicID, "dv/%i/client", this->espID);
      this->mqtt->subscribe(bufferTopicID, 0);

      yield();
                   
      //Publicamos también estado para eliminar el md:-1
      this->sendMessageStatusTopic();
             
}

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/***************************************************************
 ******* Constructor de la función callback que se ejecutará al recibir mensaje MQTT
 ****************************************************************/

std::function<void(char*, uint8_t*, unsigned int)> MQTTManager::onMessageUserCallback() {
    return [this](char* topic, byte* payload, unsigned int length) {

		/* My callback function. Not important here */
    
    };
}

`
============ CODE END ===================

Here, the "loop" method run every time runs the Arduino "loop" function. 

The other important function is the "run()" method, which runs every X seconds. 

The "run()" method cheks if we have a WiFi connection. If so, it cheks it there's a MQTT connection. If so, it sends a MQTT message.

In order to check whether there's a MQTT connection, "connect()" method is invoked. This method tries to reconnect if MQTT connection was lost.

With 2.3.0 ESP8266 core, this code can run for ages without a MQTT connection fails. If the MQTT connections is lost (I restart mosquitto, for instance), it reconnects flawlessly.

With 2.4.3 core, MQTT connetion get lost in minutes and never reconnects. Here it is what you can see in serial debug:

`
============ SERIAL OUTPUT BEGIN ===================

[MQTTManager] run() Start
18:08:02.947 -> [MQTTManager] run() WIFI Ok
18:08:02.947 -> [MQTTManager] connect() Start
18:08:02.947 -> [MQTTManager] connect() Wasn't connected
[MQTTManager] connect() Connection successful
18:08:03.513 -> [MQTTManager] run() Connected to MTTT broker
18:08:03.513 -> [MQTTManager] run() End
[MQTTManager] run() Start
18:08:31.946 -> [MQTTManager] run() WIFI Ok
18:08:31.946 -> [MQTTManager] connect() Start
18:08:31.946 -> [MQTTManager] connect() Already connected
18:08:31.946 -> [MQTTManager] run() Connected to MTTT broker
18:08:31.946 -> [MQTTManager] run() End
[MQTTManager] run() Start
18:09:00.943 -> [MQTTManager] run() WIFI Ok
18:09:00.943 -> [MQTTManager] connect() Start
18:09:00.943 -> [MQTTManager] connect() Already connected
18:09:00.943 -> [MQTTManager] run() Connected to MTTT broker
18:09:00.943 -> [MQTTManager] run() End

/* Same pattern repeats over and over until connection lost */

/* Here, last connected "run()" execution */

[MQTTManager] run() Start
18:15:17.921 -> [MQTTManager] run() WIFI Ok
18:15:17.921 -> [MQTTManager] connect() Start
18:15:17.954 -> [MQTTManager] connect() Already connected
18:15:17.954 -> [MQTTManager] run() Connected to MTTT broker
18:15:17.954 -> [MQTTManager] run() End
[MQTTManager] run() Start

/* Here, connection is lost */

18:15:46.920 -> [MQTTManager] run() WIFI Ok
18:15:46.953 -> [MQTTManager] connect() Start
18:15:46.953 -> [MQTTManager] connect() Wasn't connected
[MQTTManager] connect() Connection fail
18:15:52.036 -> [MQTTManager] run() Not MQTT connection
18:15:52.036 -> [MQTTManager] run() End
[MQTTManager] run() Start
18:16:15.952 -> [MQTTManager] run() WIFI Ok
18:16:15.952 -> [MQTTManager] connect() Start
18:16:15.952 -> [MQTTManager] connect() Wasn't connected
[MQTTManager] connect() Connection fail
18:16:20.936 -> [MQTTManager] run() Not MQTT connection
18:16:20.936 -> [MQTTManager] run() End
[MQTTManager] run() Start
18:16:44.932 -> [MQTTManager] run() WIFI Ok
18:16:44.932 -> [MQTTManager] connect() Start
18:16:44.932 -> [MQTTManager] connect() Wasn't connected
[MQTTManager] connect() Connection fail
18:16:49.947 -> [MQTTManager] run() Not MQTT connection
18:16:49.947 -> [MQTTManager] run() End

/* From here, same pattern repeats. Never reconnects*/

============ SERIAL OUTPUT END ===================

In my code I'm using this MQTT client: https://github.com/knolleary/pubsubclient/releases/tag/v2.6
I've also tryed this one with exactly the same result: https://github.com/Imroy/pubsubclient

Any ideas? 

Thanks in advance and best regards!!

ascillato, 2018-10-23:
Hi,

We are using also pubsubclient in Sonoff-Tasmota and we are not having issues with any of the cores versions. It work flawlessly no matter which lwip selected also. Please, recheck your sketch with the examples from pubsubclient and also check for your buffers. Thanks.

devyte, 2018-10-23:
A whole lot has changed bewteen 2.3.0 and 2.4.2. I suggest wiping your flash before reflashing.
In view of previous comment, and because the issue is with a 3rd party lib and  not our core, I'm closing.

closed 2018-10-23 by devyte

ikin77, 2018-10-24:
Thanks to all of you. I'll try to wipe and I'll let you know.

Best regards