esp8266/Arduino #5272: Nodemcu +I2C + deepsleep = hangs on wire.begin()

vahooz, 2018-10-24:
**Use case:**
To be able to perform the communication after returning from "deepSleep". When 20 seconds pass, and **setup()** procedure is executing nodemcu hangs on **Wire.begin()** .
**Question:** How to close  down the I2C hardware?

`#define D1 5 // I2C Bus SCL (clock)
#define D2 4 // I2C Bus SDA (data)
#include <Wire.h>
#include <VL53L1X.h>

VL53L1X sensor;

void setup()
{
  Serial.begin(19200);  
  Wire.begin(D2, D1);  
  Wire.setClock(400000); // use 400 kHz I2C  
  sensor.setTimeout(500);
  if (!sensor.init())
  {
    Serial.println("Failed to detect and initialize sensor!");
    while (1);
  }  
  sensor.setDistanceMode(VL53L1X::Long);
  sensor.setMeasurementTimingBudget(50000);
  sensor.startContinuous(50);
}

void loop()
{
  Serial.print(sensor.read());
  if (sensor.timeoutOccurred()) { Serial.print(" TIMEOUT"); }
  unsigned int sleep_secs = 20;
  ESP.deepSleep(sleep_secs*1000000, WAKE_RF_DEFAULT);
  delay(100);
}`

devyte, 2018-11-02:
Previous comment is correct: relevant info is missing, such as board, core version, etc.  Template was ignored.

Looking at the current I2C code, there is no method available in Wire, and I don't see a way to shut down in the underlying twi code to implement such a method. It would have to be done from scratch.

None of the currently active developers is involved with I2C at this time, so the enhancement would have to be picked up by a user willing to implement and contribute. @suculent ?

About your specific issue and getting stuck, waking up from deepSleep is essentially a full reset (I assume you connected the relevant pins together?), so I suspect shutting down the hardware shouldn't be needed. However, there is a known issue with deepSleep in the sdk, where the ESP sometimes doesn't wake up again, although it's supposed to happen with larger sleep values. Still, I suggest trying deepSleepInstant.

I'm keeping this issue open to track the I2C enhancement, not the deepSleep issue.

suculent, 2018-11-02:
Wire.begin() methods are re-entrant as far as I’ve experienced. It should’t get stucked first without sleep. Stacktrace would help indeed. This seems more like issue where you’d have to stop continuous mode of the Lidar first.

Tech-TX, 2019-11-23:
There is some other problem with vahooz' setup, and not enough information to see what.  I've already read the datasheet for that ST Micro lidar for use in a CPR manikin monitor I'm eventually going to make.  The sensor is an I2C slave only: as long as the I2C transaction is completed before entering DeepSleep it shouldn't present any problems.  The library is using generic Wire.begin calls, so as long as the lidar isn't insane is should complete before it returns to the user's program (they're both in CONT).  The only thing I haven't tracked down so far is the ets_task(...) call made from twi_init() in core_esp8266_si2c.cpp; that could potentially hang the Wire.begin() if the ESP isn't in a happy state after DeepSleep reset.  The rest of the twi_init() code shouldn't cause Wire.begin() to hang.

Additionally, I have a NodeMCU with a Doit.am ESP-12F module running I2C, serial, WiFi and then DeepSleep(25minutes) that's been operating continuously since August.  *Not even once* have I gotten a bad packet or a hung I2C bus.  Since I'm running a much longer DeepSleep cycle, there's a possibility that a short (20sec) DeepSleep isn't waking up correctly, but that's not an I2C issue.  A simple test of that would be to check the reset cause and hit it with ESP.restart() if it's coming out of DeepSleep.  Could be a faulty flash chip, could be the reset circuit on his ESP module, could be poor power, can't tell from here.

Tech-TX, 2019-12-30:
I think there's something odd in the beginning of his file above that's causing his problem.  I tried it here and it never got into the setup() part of the code. I added a Serial.println("Initializing...") just after the Serial.begin() that never executed.  I copied and pasted bits of my code in place of his, and it works fine.  Here's what I'm running, with an improved start-up version of what he has:
```cpp
#include <Wire.h>
#include <VL53L0X.h>

#define D1 5  // I2C Bus SCL (clock)
#define D2 4  // I2C Bus SDA (data)

VL53L0X sensor;

void setup()
{
  Serial.begin(74880);
  Wire.begin(D2, D1);
  if (millis() < 2000)  //  takes 2 seconds to boot the sensor after power is applied
    delay(2000 - millis());  // without a delay, the initialization of the sensor is erratic
  sensor.setTimeout(500);
  if (!sensor.init())
  {
    Serial.println("Failed to detect and initialize sensor!");
    while (!sensor.init()) {
      char err = Wire.status();  // clear the bus
      Serial.print("error = ");
      Serial.println(err);
      delay(50);
    }
  }

  // Start continuous back-to-back mode (take readings as
  // fast as possible).  To use continuous timed mode
  // instead, provide a desired inter-measurement period in
  // ms (e.g. sensor.startContinuous(100)).
  //sensor.setDistanceMode(VL53L1X::Long);  mine doesn't support this
  sensor.setMeasurementTimingBudget(50000);
  sensor.startContinuous(50);
}

void loop() {
  Serial.println(sensor.readRangeContinuousMillimeters());
  if (sensor.timeoutOccurred()) {
    Serial.print(" TIMEOUT");
  }
  unsigned int sleep_secs = 20;
  ESP.deepSleep(sleep_secs * 1000000, WAKE_RF_DEFAULT);
  delay(100);
}
```
I changed the serial baud rate so I could see the boot message.  Here's the monitor output:

> load 0x4010f000, len 3456, room 16 
> tail 0
> chksum 0x84
> csum 0x84
> v3197d2ac
> ~ld
> 120     << reading from the sensor
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3456, room 16 
> tail 0
> chksum 0x84
> csum 0x84
> v3197d2ac
> ~ld
> 118     << reading from the sensor
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 3456, room 16 
> tail 0
> chksum 0x84
> csum 0x84
> v3197d2ac
> ~ld
> 116     << reading from the sensor

<repeats every 22 seconds or so>

There's no problem with the Wire library or Deep Sleep, as it's been running for about 30 minutes now.  There's no need to unload the library to solve his problem, so you can close this one unless you want to consider unloading the library as an enhancement.  On the master side there's nothing to do; the master I2C doesn't _do_ anything if you don't call it.  On the slave side, we could simply disable the interrupts  which would shut the slave down until it was re-initialized.

devyte, 2019-12-30:
Thanks @Tech-TX!
Closing.

closed 2019-12-30 by devyte