esp8266/Arduino #5273: OTA - No Network port is shown !

HamzaHajeir, 2018-10-24:
Hi guys.
I've worked 2 days ago to flash my esp Over The Air.
Uploaded BasicOTA official sketch, All things went fine. Tested file uploading with success.

Today I went to develop the code and merge it in a real project. but after i upload it by serial, Network port won't show anymore (Only Serial) !

So tried back to BasicOTA sketch, with no success.

I've dived into [this topic](https://github.com/esp8266/Arduino/issues/2415) and reached a good point to the final solution.

See @krzychb's [response ](https://github.com/esp8266/Arduino/issues/2415#issuecomment-241339934), I tried a similar command below, See results : 

[image]

OTA uploading is success! but until now there's no Network port is shown at Arduino IDE.

* I've restarted Arduino IDE many times with no success.

#### Platform
- Hardware:					[ESP8266 NodeMCU 1.0 v3]
- Development Env:		[Arduino IDE 1.8.7]
- Operating System:		[Windows 10 x64]

### Settings in IDE
[image]

My esp8266's core is updated (2.4.2)

devyte, 2018-10-24:
After serial upload, did you physically reset the ESP (e.g.: press the reset button)?

HamzaHajeir, 2018-10-24:
@devyte  Yes i did many times. 

Also tried to re-power it manually by re-plugging its usb cable.

devyte, 2018-10-24:
Oh, it's windows. Did you disable the firewall, or at least set up a firewall rule? OTA works with the ESP's MDNS responder, which is UDP.

devyte, 2018-10-24:
Also, closing as this is not an issue with the core.

closed 2018-10-24 by devyte

HamzaHajeir, 2018-10-25:
Yes I've fully disabled firewall and created rules that works with active firewall.
[image]

If i did not, it wouldn't flash OTA manually.

peros550, 2018-12-16:
> Also, closing as this is not an issue with the core.

I'm having the same issue. Where should we report this issue?

devyte, 2018-12-16:
It's already fixed post beta1: ArduinoOTA now calls MDNS.update() to maintain it, assuming begin() or begin(true).

HamzaHajeir, 2018-12-17:
Hi @peros550, I've tried to search for a solution. I've opened a thread at esp8266 forums. 

But i didn't find the solution yet.

Might one approach mentioned there would work well for you.

https://www.esp8266.com/viewtopic.php?f=28&t=18712&p=79014

Please tell me if you get the solution.

devyte, 2018-12-17:
@HamzaHajeir ArduinoOTA relies on MDNS,  and MDNS was sort of broken before. Try with beta2.
Also, your laptop firewall must allow the traffic, try disabling it for testing.

HamzaHajeir, 2018-12-17:
> @HamzaHajeir ArduinoOTA relies on MDNS, and MDNS was sort of broken before. Try with beta2.
> Also, your laptop firewall must allow the traffic, try disabling it for testing.

I've reached a point all firewalls are enabled xD 

But for me all firewall settings are ok, Because i can upload binary file OTA but by a command via Command Prompt.