esp8266/Arduino #5274: Problem #include <queue.h>

Mat1dan, 2018-10-24:
Hello,
first of all the specifications of the IDE and yC are not helpful to solve the problem....
So i have copied the libary ESP8266WiFi to .....\Arduino\libaries.
After compiling my project i've got following error:

`Build options changed, rebuilding all
In file included from C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:33:0,
 from C:\(...)\Snoware20http1.1\ESP_Webserver_HTTP11\ESP_Webserver_HTTP11.ino:6:
C:\Program Files (x86)\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiType.h:26:19: fatal error: queue.h: No such file or directory
 #include <queue.h>
      ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Uno.
`
Where is <queue.h> and where should this file be located?

Thank you!

bertmelis, 2018-10-24:
I would think your IDE does matter to solve your problem. queue.h is part of the SDK. (look in 'include' directory).

Why did you copy the lib?

Mat1dan, 2018-10-24:
Ohhh thank you...
It was my fault! 
However i have an other error now
`Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, QIO, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

ESP_Webserver_HTTP11:9: error: conflicting declaration 'SoftwareSerial Serial1'

 SoftwareSerial Serial1(6, 7); // Virtuelles RX, TX

                       ^

In file included from (...)\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Arduino.h:261:0,

                 from sketch\ESP_Webserver_HTTP11.ino.cpp:1:

(...)\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/HardwareSerial.h:200:23: error: 'Serial1' has a previous declaration as 'HardwareSerial Serial1'

 extern HardwareSerial Serial1;

                       ^

exit status 1
conflicting declaration 'SoftwareSerial Serial1'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
Where is the sense here XD

devyte, 2018-10-24:
The description hints at wrong installation of the core, and the error messages agree.
Closing.

closed 2018-10-24 by devyte

bertmelis, 2018-10-25:
For the record, the error message does make sense. Did you actually read it? Do you know basic C++?

You cannot call your softserial 'Serial1' because that name is already taken. Pick something else.

miguerman, 2018-11-17:
Hi.
"ESP8266WiFiType.h is referencing queue.h"
"C: \ Program Files (x86) \ Arduino \ libraries \ ESP8266WiFi \ src / ESP8266WiFiType.h: 26: 19: fatal error: queue.h: No such file or
How is the problem solved?

pabatribick, 2018-12-12:
I've the same problem with the "queue.h" file. No one has an solution?

Mat1dan, 2018-12-12:
I deleted ESP8266.h and all the other files which belonged to this Project. I am using only the lib from the Arduino IDE http://arduino.esp8266.com/versions/2.4.2/package_esp8266com_index.json

devyte, 2018-12-12:
As stated above, queue.h is part of the SDK and not part of the core. The error hints at wrong installation. 
I suggest deleting the entire core and IDE and installing from scratch. Remember to restart your IDE after installing. Also, if you're installing latest git, remember to `cd tools ; ./get.py`.

Do _not_ install a core via board manager and then try to update some core libs manually. It might work, or it might not. Or it might compile and crash the ESP. Or it might cause your neighbor's TV to change channel. That's unlikely, but please let me know if it happens.

diraniyoussef, 2020-03-15:
I have the same problem, but in eclipse. Where to add this link 
http://arduino.esp8266.com/stable/package_esp8266com_index.json
in eclipse ?
I followed the steps to have Arduino Downloads Manager, but I get this queue.h error.

Mat1dan, 2020-03-15:
Did you already try the installation guide behind following link? https://arduino-esp8266.readthedocs.io/en/latest/installing.html

diraniyoussef, 2020-03-17:
My problem is in eclipse. not in Arduino IDE. 
For a beginner like me, I went to https://docs.platformio.org/en/latest/integration/ide/eclipse.html and I didn't understand well. Then I went to https://arduino-esp8266.readthedocs.io/en/latest/eclipse/eclipse.html and I slightly understood something, but not the thing below on how to fix. Sorry.
Then I may try now Sloeber since the Arduino and esp8266 can be automated. I guess ?

LinuxRocks2000, 2021-01-21:
> For the record, the error message does make sense. Did you actually read it? Do you know basic C++?
> 
> You cannot call your softserial 'Serial1' because that name is already taken. Pick something else.

"Do you know basic C++" is a bit harsh. Whether or not the OP uses C++ is not related to the question, he is asking about *usage of a library*.

bertmelis, 2021-01-21:
???

OP had an error:
error: 'Serial1' has a previous declaration as 'HardwareSerial Serial1'
If you read and understand C++ it easy to solve.

But next time I'll try to be less "stackoverflow".