esp8266/Arduino #5275: Crash on     SPIFFS.format()    or   SPIFFS.begin()    only with ESP-12s.

javiercuellar73, 2018-10-24:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12s
- Core Version:			2.4.2
- Development Env:		Arduino IDE  1.8.7
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Generic ESP8266 Module|
- Flash Mode:			DIO
- Flash Size: 		4MB
- lwip Variant:    v1.4 Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I have several ESP-12F and ESP-12S.
The sketch works fine on ESP-12F but crash on ESP-12S (I have testes at least 5 different units.
Problem is similar to the one reported on [4061](https://github.com/esp8266/Arduino/issues/4061)
Attached a picture of flash, just in case is same problem.

On instruction SPIFFS.begin(); it crash.
Any ideas?

Thanks!!!

```
#include "FS.h"

String realSize = String(ESP.getFlashChipRealSize());
String ideSize = String(ESP.getFlashChipSize());
bool flashCorrectlyConfigured = realSize.equals(ideSize);

void setup() {
Serial.begin(9600);

Serial.println("Starting...");
if(flashCorrectlyConfigured){
Serial.println("flash correctly configured, SPIFFS started, IDE size: " + ideSize + ", real size: " + realSize);
SPIFFS.begin();
Serial.println("After begin...");
}
}

void loop() {

}
```

### Debug Messages

```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:1.4.0rc2/BearSSL:6d1cefc
Starting...
flash correctly configured, SPIFFS started, IDE size: 4194304, real size: 4194304
SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @100000, size=2fb000, block=2000, page=100
SPIFFSImpl: mount rc=-10025
Fatal exception 0(IllegalInstructionCause):
epc1=0x4022dc70, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x4022dc70 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffffad0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffc70:  feefeffe feefeffe feefeffe feefeffe  
3ffffc80:  00000000 00000000 00000020 40101232  
3ffffc90:  3ffe95b0 40104614 3ffefe0c feefeffe  
3ffffca0:  00000001 40103833 3ffed730 feefeffe  
3ffffcb0:  40103c96 feefeffe feefeffe feefeffe  
3ffffcc0:  40104965 40104962 7fffffff 00000000  
3ffffcd0:  400005e1 0000007f 7fffffff 00000000  
3ffffce0:  4022dc70 00000033 00000016 40104f0d  
3ffffcf0:  4010496b 04000102 00000000 00000001  
3ffffd00:  fbf8ffff 04000002 3feffe00 00000100  
3ffffd10:  0000001a 00000018 04000102 4010494c  
3ffffd20:  3fffc100 00101000 00000000 00000000  
3ffffd30:  00007fff 0004a72d 3ffeda48 4010211e  
3ffffd40:  3ffe9e3c 00000000 00000000 0000001c  
....
....
....
```

[image]
[image]

closed 2018-10-24 by javiercuellar73

faparo, 2019-03-06:
Hi Javier. Why you closed this issue? Do you find the problem and fix it?
I have the same problem but i can´t fix it. Can you help me?

javiercuellar73, 2019-03-06:
Hi

I closed it because after several tests, it was working, but sometimes it
fails.
Because I just wanted to save a variable, I change and save it to EPROM.