esp8266/Arduino #5277: Add a function for insecure HTTPS connections with HTTPClient

N-Storm, 2018-10-25:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					Any ESP
- Core Version:			latest git
- Development Env:		Arduino IDE
- Operating System:		GNU/Linux (Ubuntu variant)

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			qio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		board
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

BearSSL allows for 'insecure' TLS connections by calling `WiFiClientSecure::setInsecure()` function. It turns off certificate and/or fingerprint checking which is useful in some cases.
Currently HTTPClient doesn't allows such connections because it lacks methods to set 'insecure' connection. While there was some argumentation against insecure connections through (already closed) issues they were based on axTLS lib and also I think this aren't practical not to include this for the following reason.
Sometimes you want to have a connection initiated with some sort of public HTTPS server which you don't have control over. And which can change it's TLS cert any time or even on regular basis. It's not feasible to include every trusted CA root certificate like programs on bigger devices do to handle this. But sometimes you don't care about possible MITM attack. All you need is just to grab some data over HTTPS from 3rd party source if it's not available over plain HTTP.
It's very simply to add additional overloaded function `HTTPClient::begin()`, where 2nd argument could be of bool value for example and take false to initiate insecure connection.
I've did a [workaround for me as a class extension](https://gist.github.com/N-Storm/e0e02592affc33cfba35412ff298d770). It works but I'm not familiar with C++ well enough, a link are just for a reference here.
Probably a more convenient feature would be to parse protocol on HTTPClient::begin() when only 1 argument are supplied and initiate a insecure HTTPS connection if the url begins with `https://`.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>

ESP8266WiFiMulti WiFiMulti;

void setup() {
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("SSID", "PASSWORD");

  while ((WiFiMulti.run() != WL_CONNECTED)) {
    delay(500);
  }

  HTTPClient http;

  http.begin("https://github.com", false);
  int httpCode = http.GET();
  Serial.print("[HTTP] Resonpose code: ");
  Serial.println(httpCode, DEC);
  http.end();
}

void loop() {
}```

Jeroen88, 2018-10-25:
You could also use the new begin() methods that pass in a WiFiClient(Secure). See BasicHttpsClient.ino for an example and use a WiFiClient or a BearSSL::WiFiClientSecure and use client -> setInsecure() instead of client -> setFingerprint(). Next call a URL that begins with http:// instead of http://

devyte, 2018-10-25:
Previous comment is correct. Create a standalone WiFiClient or WiFiClientServer, configure it, then pass it as argument to begin().

closed 2018-10-25 by devyte

coder137, 2019-01-16:
Hey, I found this thread when facing the same issue.

When using `BearSSL::WifiClientSecure`
I get code -1 with `https.get`

``` c
HTTPClient https;
BearSSL::WiFiClientSecure newSecure;
newSecure.setInsecure();
int checkBegin = https.begin(newSecure, "raw.githubusercontent.com", 443, "/coder137/TM4C123G/master/.gitattributes", false);
Serial.println(checkBegin);

int code = https.GET();
String payload = https.getString();
Serial.println(code);
Serial.println(payload);

// !
https.end();
newSecure.stop();
```

However when using the default axTLS::WifiSecure
This works perfectly

``` c
HTTPClient https;
WiFiClientSecure newSecure;
int checkBegin = https.begin(newSecure, "raw.githubusercontent.com", 443, "/coder137/TM4C123G/master/.gitattributes", false);
Serial.println(checkBegin);

int code = https.GET();
String payload = https.getString();
Serial.println(code);
Serial.println(payload);

// !
https.end();
newSecure.stop();
```

However in the docs it has been written that BearSSL is preferred over axTLS since the latter is deprecated. 
Am I doing something wrong or is this a bug?