esp8266/Arduino #5281: Sketch won’t upload when sensor is connected to D7/D8

FutureTense, 2018-10-27:
I’m trying to use the USB port for power and uploading sketches and want to use a sensor for rx/tx serial transmissions.  I’ve hooked my breadboard up and connected D7 and D8 to my sensors rx/tx pins and with an empty sketch I can compile but not upload.

When I remove the sensor I can upload.  It seems the D7 and D8 pins are probably interfering with the serial upload.  Is this expected behavior?  I was planning on using serial.swap to use the sensor, but having to remove the component to upload doesn’t seem like an ideal situation.

Pablo2048, 2018-10-27:
if D8 == GPIO15 => read https://github.com/esp8266/Arduino/blob/master/doc/boards.rst#boot-messages-and-modes

devyte, 2018-10-27:
Previous comment is correct. The GPIO15 state at boot time affects boot mode.
Closing

closed 2018-10-27 by devyte