esp8266/Arduino #5285: Binary images have debug information on Arduino for Windows

sreekants, 2018-10-28:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I was looking at optimizing file size. I did a strings command on the binary image (XXX.ino.bin) that Arduino generates. There are a lot of unnecessary strings in the binary image that I think ought to be stripped out. It simply wastes space.

==Issue 1==
I see a lot of strings in there that should be assertions used only for testing. The use of uassert() in lwip-esp.c & lwip-git.c seems to generate the following strings in the image. Shouldn't the assertions be conditional?

assert fail: p->pbuf.type == PBUF_CUSTOM_TYPE_POOLED @%s:%d
glue-esp/lwip-esp.c
assert fail: p->pbuf.flags == 0 @%s:%d
assert fail: p->pbuf.next == NULL @%s:%d
assert fail: p->pbuf.eb == NULL @%s:%d
assert fail: netif == test_netif_sta || netif == test_netif_ap @%s:%d
assert fail: packet_incoming == ethernet_input @%s:%d
assert fail: p->ref == 1 @%s:%d
assert fail: !p->eb @%s:%d
assert fail: pw->ref2save @%s:%d
assert fail: netif->num == 0 || netif->num == 1 @%s:%d
assert fail: p->tot_len == p->len && p->ref == 1 @%s:%d
assert fail: netif->num == STATION_IF || netif->num == SOFTAP_IF @%s:%d
glue-lwip/lwip-git.c
 

==Issue 2==
I am also not sure why the following strings to the path of the package are in the binary image.
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/DataSource.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/DataSource.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/DataSource.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/DataSource.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/ClientContext.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/ClientContext.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/ClientContext.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/ClientContext.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/DataSource.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src\include/DataSource.h
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\abi.cpp
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\core_esp8266_main.cpp
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\spiffs_api.h

It seems like a lot of files paths are built into the binary image unnecessarily when you include a feature. Are the build settings for the cross compiler optimized to eliminate these kind of debug information? I wonder.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

The standard builtin WiFiClient is a sufficient test case.

### Debug Messages

NA.

dragondaud, 2018-10-29:
This might explain a change I noticed recently, which causes problems in [one of my projects](https://github.com/dragondaud/myClock). If I compile on Windows or MacOS the resultant image has 2500 or so bytes less free heap than the exact same code built on Linux. All using latest Arduino IDE 1.8.7 and same libraries. The result is myClock built under Windows runs out of heap when serving web pages, but works fine when built on Linux. I have started supplying linux-built binaries as a work-around, but that doesn't allow my users to customize the code for their needs, unless they use Linux.

earlephilhower, 2018-10-30:
Assertions have the full file path in the text, so that would explain any differences in sizes between Linux-built and Windows-built.  Check the `Tools->Debug Level->Np Assert/NDEBUG`option to strip them completely away.

earlephilhower, 2018-10-30:
The original listed assert strings seem to be coming from LWIP.  Might be some macro issues w/the LWIP compilation step.

I've recently been burned trying to boot up my Win VM, so would you be able to post a WiFiClient build `.elf` (not .bin) and `.map` file so I can do some looking at them?

Thx

dragondaud, 2018-10-30:
Here you go
[WiFiClient-build.zip](https://github.com/esp8266/Arduino/files/2531050/WiFiClient-build.zip)

dragondaud, 2018-10-30:
I hadn't directly compared such a basic build between Win & Linux. Only seeing 492 bytes less on Windows in that case, so I dug thru my own elf, and discovered the culprit was actually WiFiManager. On my Linux system I was tracking 'development' and on WIndows and MacOS I was using the default master. Using the 'NoAssert-NDEBUG' is probably the answer to the original question, and my issue was unrelated. Sorry :)

devyte, 2018-10-30:
@sreekants can you please confirm that 
> Check the Tools->Debug Level->Np Assert/NDEBUG option

addresses your issue?

sreekants, 2018-10-31:
No it did not address the issue. I still see the assertions.  It did however reduce the number of strings that came from the DataSource.h file on Issue 1.

==Issue 1==
These still remain on 
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\abi.cpp
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\core_esp8266_main.cpp
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\spiffs_api.h

==Issue 2==
No change.

Here's something that might help... I greped the files installed in my packages. One of the line in question seems to the following.

C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2\builder\glue-lwip\lwip-git.c(235,10): 	uassert(netif->num == STATION_IF || netif->num == SOFTAP_IF);

If you further grep the usassert macro,  There are two definitions

---------- Find in Files ----------
> Searching for the string 'uassert'...
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2\builder\glue\gluedebug.h(85,9): #define uassert(assertion...)	do { if ((assertion) == 0) { os_printf_plus("assert fail: " #assertion " @%s:%d\n", __FILE__, __LINE__); uhalt(); } } while (0)
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\tools\sdk\lwip2\include\gluedebug.h(85,9): #define uassert(assertion...)	do { if ((assertion) == 0) { os_printf_plus("assert fail: " #assertion " @%s:%d\n", __FILE__, __LINE__); uhalt(); } } while (0)
> 2 occurrence(s) have been found.

My sense is that assertion macros ought to be wrapped around #ifdef. But it does not seem to be the case. Perhaps that is the problem.

#define uassert(assertion...)	do { if ((assertion) == 0) { os_printf_plus("assert fail: " #assertion " @%s:%d\n", __FILE__, __LINE__); uhalt(); } } while (0)

sreekants, 2018-10-31:
> @sreekants
>
> All those asserts (uassert) are lwip2's only, and I checked both core-2.4.2 and git master and they are not present in tools/sdk/lib/liblwip2*.a (you found the sources but they are disabled in binaries).
> So either you are running an older version of the core (in which those asserts were still enabled), or you did recompile lwip2 with those asserts enabled.
>
> Can you print ESP.getFullVersion() in your sketch ?

Is this what you are looking for?
Serial.println( ESP.getCoreVersion() );
Serial.println( ESP.getSdkVersion() );

2_4_0
2.1.0(deb1901)

d-a-v, 2018-10-31:
> Is this what you are looking for?
> 2_4_0

Yes. This issue will disappear with v2.4.2 which is the latest release.

sreekants, 2018-11-02:
Kudos to the team. I just upgraded to 2.4.2. The size of global variables went down from 69% to 53% with that single change. 

I am exploring a little further now. I'll keep you posted of new findings.

sreekants, 2018-11-02:
Almost everything disappears when we the  'NoAssert-NDEBUG'  option is selected as suggested by @dragondaud . This further reduced the size of the global variables down to 51%.

@d-a-v ,  the only other strings I see in the image that I think can be resolved right away is the following when I do a 'strings' on my image sensoros.ino.bin:

....
MD5 Failed: expected:%s, calculated:%s
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\abi.cpp
__throw_bad_alloc
__throw_bad_function_call
__cxa_pure_virtual
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\core_esp8266_main.cpp
2_4_2
loop_task
__yield
C:\Users\blue__000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\spiffs_api.h

I don't see a reason for why the path to these files ought to be in the binary when I build it with 'NoAssert-NDEBUG' . Any suggestions there?