esp8266/Arduino #5287: Compilation fails with recent Wire.h changes

mcspr, 2018-10-28:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				Any
- Core Version: https://github.com/esp8266/Arduino/commit/cb05b86d49717ce155db618b9cb9e9cc18ced92f
- Development Env:		Arduino IDE / Makefile
- Operating System:		Linux

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Commit https://github.com/esp8266/Arduino/pull/5226/commits/bfcbd71eb924fc6cdd38d79c4c8869542afc2f4a from #5226  changed `twi_attachSlaveRxEvent` second argument to `size_t` and now when `Wire.h` is included compilation fails.

Looking at [arduino-avr](https://github.com/arduino/ArduinoCore-avr/blob/2663be17272e19f00c55f3f2d8f1ebfac47158d6/libraries/Wire/src/utility/twi.h#L48) variant, [`cores/esp8266/twi.h`](https://github.com/esp8266/Arduino/blob/cb05b86d49717ce155db618b9cb9e9cc18ced92f/cores/esp8266/twi.h) and [`cores/esp866/core_esp8266_si2c.c`](https://github.com/esp8266/Arduino/blob/cb05b86d49717ce155db618b9cb9e9cc18ced92f/cores/esp8266/core_esp8266_si2c.c) seems to be in the wrong here and 'numBytes' should be `int` instead of `size_t`
https://github.com/esp8266/Arduino/blob/cb05b86d49717ce155db618b9cb9e9cc18ced92f/libraries/Wire/Wire.cpp#L215
https://github.com/esp8266/Arduino/blob/cb05b86d49717ce155db618b9cb9e9cc18ced92f/cores/esp8266/twi.h#L51
https://github.com/esp8266/Arduino/blob/cb05b86d49717ce155db618b9cb9e9cc18ced92f/libraries/Wire/Wire.cpp#L84

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <Wire.h> // <---

void setup() {}

void loop() {}

```

### Debug Messages

```
/home/builder/arduino-1.8.7/hardware/esp8266com/esp8266/libraries/Wire/Wire.cpp: In member function 'void TwoWire::begin(uint8_t)':
/home/builder/arduino-1.8.7/hardware/esp8266com/esp8266/libraries/Wire/Wire.cpp:84:42: error: invalid conversion from 'void (*)(uint8_t*, int) {aka void (*)(unsigned char*, int)}' to 'void (*)(uint8_t*, size_t) {aka void (*)(unsigned char*, unsigned int)}' [-fpermissive]
   twi_attachSlaveRxEvent(onReceiveService);
                                          ^
In file included from /home/builder/arduino-1.8.7/hardware/esp8266com/esp8266/libraries/Wire/Wire.cpp:31:0:
/home/builder/arduino-1.8.7/hardware/esp8266com/esp8266/cores/esp8266/twi.h:51:6: error:   initializing argument 1 of 'void twi_attachSlaveRxEvent(void (*)(uint8_t*, size_t))' [-fpermissive]
 void twi_attachSlaveRxEvent( void (*)(uint8_t*, size_t) );
      ^
===info ||| Using library {0} at version {1} in folder: {2} {3} ||| [Wire 1.0 %2Fhome%2Fbuilder%2Farduino-1.8.7%2Fhardware%2Fesp8266com%2Fesp8266%2Flibraries%2FWire ]
exit status 1
```

mcspr, 2018-10-28:
Also mentioned in https://github.com/esp8266/Arduino/commit/cb05b86d49717ce155db618b9cb9e9cc18ced92f#comments, but suggested change is other way around - fix other args to `size_t`

ascillato, 2018-10-28:
Hi,

I can confirm that the core is not compiling if using the core from https://github.com/esp8266/Arduino/commit/cb05b86d49717ce155db618b9cb9e9cc18ced92f and this wire library in a sketch. So, for example, [Sonoff-Tasmota](https://github.com/arendst/Sonoff-Tasmota) will not compile either under this commit.

Also I can confirm that the fix proposed in the [comments ](https://github.com/esp8266/Arduino/commit/cb05b86d49717ce155db618b9cb9e9cc18ced92f#commitcomment-31072583) don't work either. The Arduino core stills not compile. Sorry.

Please, if there is something else I can help, do not hesitate on asking.

devyte, 2018-10-28:
The correct arg type is in fact size_t, and not int, because the value can't be unsigned. At the very least, it needs to be unsigned.
The proposed fix is in #5289 , and it compiles for me. If the core doesn't compile for you, I'll need further details.

ascillato, 2018-10-28:
Hi, 

I can confirm that with the PR #5289, now the core compiles fine.

Please, merge it.

Thanks a lot for your work :+1:

PR #5289 (2018-10-28): Fix arg type in Wire to size_t
Fixes #5287
closed 2018-10-28 by devyte