esp8266/Arduino #5298: NTP-TZ-DST.ino

Igorystiy, 2018-10-30:
When I use this example, the Serial displays 
 date and time. PTM(year) and PTM (mon) not correct. Other PTM data is displayed correctly. When using ctime(&now) or "strftime", all data, including year and month are displayed in the correct line. Is it bug?
Arduino IDE 1.8.5
ESP 8266 Community 2.4.2
ESP 12 Module.
NTP:
localtime:isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=8
...
time:28808 ctime:(UTC+120mn)Thu Jan  1 08:00:08 1970

after given UTC time:
localtime:isdst=0:yday=302:wday=2:year=118:mon=9:mday=30:hour=15:min=52:sec=14
...
time:1540914734 ctime:(UTC+120mn)Tue Oct 30 15:52:14 2018

Have idea?

TKaluza, 2018-11-06:
Thank you for #clarification.

Could it be that:

- local and gmtime are the same? 
- Daylight saving time implentation is wrong?

I'm in Germany (so TZ = 1 and DST_MN = 60). The time should be something with 22:10.
The result is: 
```
localtime:isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=2
gmtime   :isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=2

clock:2/54712000ns millis:2054 micros:2054731 gtod:2/30344us time:28802 ctime:(UTC+120mn)Thu Jan  1 08:00:02 1970
------------------ settimeofday() was called ------------------

localtime:isdst=0:yday=309:wday=2:year=118:mon=10:mday=6:hour=23:min=3:sec=59
gmtime   :isdst=0:yday=309:wday=2:year=118:mon=10:mday=6:hour=23:min=3:sec=59

clock:3/66632000ns millis:3066 micros:3066648 gtod:1541538240/903398us time:1541545439 ctime:(UTC+120mn)Tue Nov  6 23:03:59 2018
```

When using TZ 1 and DST_MN 0 I'm getting the right time;
```
localtime:isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=2
gmtime   :isdst=0:yday=0:wday=4:year=70:mon=0:mday=1:hour=8:min=0:sec=2

clock:7/742692000ns millis:7742 micros:7742713 gtod:2/30217us time:28802 ctime:(UTC+60mn)Thu Jan  1 08:00:02 1970
------------------ settimeofday() was called ------------------

localtime:isdst=0:yday=309:wday=2:year=118:mon=10:mday=6:hour=22:min=13:sec=14
gmtime   :isdst=0:yday=309:wday=2:year=118:mon=10:mday=6:hour=22:min=13:sec=14

clock:8/754610000ns millis:8754 micros:8754626 gtod:1541538795/20171us time:1541542394 ctime:(UTC+60mn)Tue Nov  6 22:13:14 2018

```

devyte, 2019-11-09:
@d-a-v can this be closed? The new TZ, NTP, etc code seems to work correctly.

d-a-v, 2019-11-09:
@TKaluza please try the new ntp-tz-dst example sketch.
There is a constant: `TZ_Europe_Berlin` for Germany.

@Igorystiy please find the major city near you in [this file](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/TZ.h), use it in the new ntp-tz-dst example.

Closing this issue, please open a new one if the new example doesn't work with you.

closed 2019-11-09 by d-a-v