esp8266/Arduino #5299: tone() crashes with frequency of zero

RichyT, 2018-10-30:
The ToneMelody sketch has this arrangement of notes which produces the "Shave and a haircut. Next please" melody. I connected it to GPIO5. When the sketch was uploaded, it played the "Shave and a haircut" part and then stopped. When reset, it would play it repeatedly over and over. but never the "Next please" part. Adding some debugging showed that loop() was never being reached. The melody is defined in the following lines...

`int melody[] = {
  NOTE_C4, NOTE_G3, NOTE_G3, NOTE_A3, NOTE_G3, 0, NOTE_B3, NOTE_C4
};`

It was failing with the 0 value. Replacing this with another note or a low value caused it to behave correctly (although the relevant note was then played instead of no note, of course).

Obviously this is not hard to work-around but if Arduino compatibility is desired, this should probably be fixed (I am assuming this works fine on an Arduino though, to be honest).

devyte, 2018-11-02:
@earlephilhower can you reproduce with latest git? If not, I suggest closing due to lack of info (issue template ignored).

RichyT, 2018-11-05:
Sorry, didn't notice issue template. I'll try giving it a go with a newer version sometime but happy to leave closed.