esp8266/Arduino #5300: HardwareSerial is missing a method to set the baud rate

tve, 2018-10-31:
I need to change the baud rate on a HardwareSerial port and I notice that there is no method to do so. Uart.h/uart.cpp has uart_set_baudrate but it's not exposed through HardwareSerial. Is this on purpose or an oversight? Using something like Serial.begin(new_baudrate) has a pile of negative side-effects and is pretty darn expensive, including calls to free and malloc. If a PR would be accepted I can prep one.

d-a-v, 2018-10-31:
PRs are always welcome !

devyte, 2018-11-02:
Having the baudrate set at init time, and not settable later, is on purpose. There are possible implications in changing later, such as receiving or sending garbage bytes, if not done carefully.
I'm curious what is the use case?

tve, 2018-11-02:
The use case is programming an AVR (optiboot) and switching baud rates to do so. Often the baud rate used by a sketch is 115200 and the esp8266 acts as transparent serial-TCP bridge, then to program the AVR from the esp8266 the baud rate hard-coded into optiboot must be used, which is often 57600 baud. The library I'm using tries to sync at the various common optiboot baud rates, i.e., it needs to switch from one to the next.
You mention "possible implications" and I'd counter that the implications of calling begin for each baud rate change are much more drastic ;-). Right now I'm working around the issue by accessing the hardware registers directly...

devyte, 2018-11-02:
What are the implications of calling begin(), besides the malloc/free?
In the meantime, please go ahead with a PR, we can discuss code details there.

tve, 2018-11-02:
You're right that other than the tx pin transitioning through being an input nothing else is disrupted.
I just noticed that it looks like HardwareSerial doesn't support swapping the rx pin, that's another function I need.

devyte, 2019-11-11:
There is already a method called updateBaudRate().
Closing.

closed 2019-11-11 by devyte