esp8266/Arduino #5303: How to trrigger serial interrupts in ESP8266?

mta9896, 2018-11-01:
Hi. I need to trigger an interrupt every time a new character is received in ESP8266. I need to be able to read rx buffer whenever a new character is received. I'm using HardwareSerial library.

I figured I should attach an interrupt routine to ESP8266 using attachInterrupt() function, the question is, which pin on ESP8266 should I attach interrupt to? I used "0" as the first argument but apparently it's not working.

Here's my code: 

`#include <HardwareSerial.h>

char input;

void handleRxInterrupt()
{
  input = Serial.read();
}

void setup()
{
  Serial.begin(115200);
  attachInterrupt(0, handleRxInterrupt, CHANGE);
}

void loop() {
}`

Also, how can I access rx buffer to save the incoming data? I used Serial.read and I don't know if that's fine.

d-a-v, 2018-11-01:
Interrupts are already handled in `uart.c` which `HardwareSerial.cpp` uses.
Bytes are stored into an internal buffer (default size of 256, can be changed) that is accessible through the standard arduino API (`Serial.available`, `.read`).

> Also, how can I access rx buffer to save the incoming data? I used Serial.read and I don't know if that's fine.

It is. `.read()` returns -1 when no data are available, use `Serial.available()` to know whether you can read.

devyte, 2018-11-01:
There is a hardware fifo that stores incoming bytes and interrupts at a certain point firing the ISR. The ISR copies the bytes from the fifo to a mem buffer. You can then read from that buffer. You don't need to worry about the details, just about reading from that mem buffer, as explained above.
If you need a different behavior, you'll have to do your own implementation. In that case you can take a look at uart.c/h for guidance.
Closing.

closed 2018-11-01 by devyte