esp8266/Arduino #5306: Pin not "writable" after tone()

fabiuz7, 2018-11-04:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [X ] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-01
- Core Version:			8ae0746e4aeaf7c2a8881831f370b40347e47a50
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini v3.0.0
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		?
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

I'm using tone(..) function to play some melodies on a passive buzzer (this is my breakout:  https://robotdyn.com/buzzer-module.html). After the melody, I should standby the buzzer to save energy, so I have to write 1 on the digital pin. I observed that after the tone function is impossible to command that pin, it stays low. The problem is not hardware, since this behaviour is observable even without a buzzer...

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
  pinMode(D6,OUTPUT);
  digitalWrite(D6, HIGH);  // the pin really goes HIGH
  delay(5000);

  tone(D6,1000,500);
  delay(1500);
  noTone(D6);   // This makes no difference, I was thinking it is necessary to "release" the pin...
  
  pinMode(D6,OUTPUT);
  digitalWrite(D6, HIGH);  // It stays low
}

void loop() {

}

```

liebman, 2018-11-04:
I can confirm this.  I modified the sketch to add a second pin to trigger a scope. The pin actually does go high but is then driven low less than 1ms later, I've seen 100us to 750us. I also tried adding a delay after ```noTone()``` to see if it was some delayed interrupt/timer but that made no difference.

```c++
#include <Arduino.h>

void setup() {
    pinMode(D5, OUTPUT);
    digitalWrite(D5, LOW); // Second scope channel used as trigger

    pinMode(D6,OUTPUT);
    digitalWrite(D6, HIGH);  // the pin really goes HIGH
    delay(5000);

    tone(D6,1000,500);
    delay(1500);
    noTone(D6);   // This makes no difference, I was thinking it is necessary to "release" the pin...

    delay(1500);
    
    pinMode(D6,OUTPUT);
    digitalWrite(D5, HIGH);  // trigger scope
    digitalWrite(D6, HIGH);  // pulses high and is changed low again after 750us
}

void loop() {

}

```

earlephilhower, 2018-11-04:
Thanks for the clear MCVE.  The ISR was disabling the pin and writing 0 every time it was called, even if it was already disabled.  Check the PR if you can, but I've verified it on my own logic analyzer.

fabiuz7, 2018-11-05:
Yeah the fix works for me, thanks!

I'm new with this procedure (pull request, complex versioning and so on...) but I would ask if this fix will be included in version 2.5.

closed 2018-11-05 by devyte

devyte, 2018-11-05:
PR is merged.

PR #5308 (2018-11-05): Only stop waveforms actually running in ISR
The ISR could end up writing a 0 to a GPIO that had previously been
stopped, effectively overwriting user writes to those pins.  Fix to
only actually disable and write when a pin was enabled and times out.

Fixes #5306
commit 6e0c0e3d (2018-11-05): Only stop waveforms actually running in ISR (#5308)

The ISR could end up writing a 0 to a GPIO that had previously been
stopped, effectively overwriting user writes to those pins.  Fix to
only actually disable and write when a pin was enabled and times out.

Fixes #5306