esp8266/Arduino #5310: How display POST request from ESP32 on webserver

KevinKHH, 2018-11-05:
Hi,

I have a NGINX server on a raspberry pi. I want to send JSON file from my ESP32 board to the NGINX web-server.

I use this code I found on a tutorial:
```
#include <WiFi.h>
#include <HTTPClient.h>
 
const char* ssid = "freebox_FSVWVO";
const char* password = "legendaireinvincible1234";
 
void setup() {
 
  Serial.begin(115200);
  delay(4000);   //Delay needed before calling the WiFi.begin
 
  WiFi.begin(ssid, password); 
 
  while (WiFi.status() != WL_CONNECTED) { //Check for the connection
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.println("Connected to the WiFi network");
 
}
 
void loop() {
 
 if(WiFi.status()== WL_CONNECTED){   //Check WiFi connection status
 
   HTTPClient http;   
 
   http.begin("http://192.168.0.45:151/postfile/receive.php");  //Specify destination for HTTP request
   //http.begin("http://jsonplaceholder.typicode.com/posts");  //Specify destination for HTTP request
   http.addHeader("Content-Type", "text/plain");             //Specify content-type header
 
   int httpResponseCode = http.POST("test=POSTING from ESP32");   //Send the actual POST request
 
   if(httpResponseCode>0){
 
    String response = http.getString();                       //Get the response to the request

    Serial.println("Return code");
    Serial.println(httpResponseCode);   //Print return code
    Serial.println("Request answer");
    Serial.println(response);           //Print request answer
 
   }else{
 
    Serial.print("Error on sending POST: ");
    Serial.println(httpResponseCode);
 
   }
 
   http.end();  //Free resources
 
 }else{
 
    Serial.println("Error in WiFi connection");   
 
 }
 
  delay(10000);  //Send a request every 10 seconds
 
}
```

When I replace the URL with my webserver one, I don't know how to display the POST request on my webserver. Could you explain me ?

Thank you,

dragondaud, 2018-11-05:
This is an issue board, not a help forum. Please close ticket and refer to esp8266.com or such for help.

devyte, 2018-11-06:
Also, this is the issue tracker for the esp8266, not the esp32.
Closing.

closed 2018-11-06 by devyte

KevinKHH, 2018-11-06:
Sorry and thank you both of you.