esp8266/Arduino #5314: Unable to connect to ESP8266 HTTP Server from Android

nick-atmosphereiot, 2018-11-06:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			6e0c0e3
- Development Env:		Make
- Operating System:		Ubuntu

### Settings in IDE

- Module:  			NodeMCU
- Flash Mode:			other
- Flash Size: 		4MB
- lwip Variant:   unknown
- Reset Method:		unknown
- Flash Frequency:	[40Mhz]
- CPU Frequency:		unknown
- Upload Using:		serial
- Upload Speed:   115200

### Problem Description

Running a Soft AP on the ESP8266, my android device is unable to communicate. It connects to the AP just fine, but all GET requests time out. The same request made on my desktop (Windows) or Laptop(Ubuntu) work just fine. I cannot even ping the device on Android.

apicquot, 2018-11-10:
It is an android issue: when connecting to a AP with no internet, Android uses in priority the cellular network that of course can not resolve a local IP. Disable the cellular data and retry. It should work.

nick-atmosphereiot, 2018-11-11:
@apicquot that makes sense, but it's very unfortunate. We are trying to work on an application that automatically connects to the ESP8266 Soft AP, and does some GET/PULL requests on it. If the user has to manually turn data off, that kind of renders the whole thing useless. 

Is there any way around this from the ESP8266 side?

apicquot, 2018-11-11:
if you develop the Android App yourself, you can restrict programmatically the use of cellular data while you connect the AP. If you are using the Android API28 you will also have to explicitly authorize http request as by default only https is authorized.

nick-atmosphereiot, 2018-11-11:
We are developing via cordova/phonegap, so I don't think we have access to the raw android API like that. I'm not the app dev guy, though, so I'm not 100% on that. 

When you connect to the wifi or something at starbucks, it forces you into a captive portal until you agree/sign in. At that point, you don't really have "internet", you are just connecting to a local page. That works with android devices, so maybe something similar can be done on the ESP8266 side.

devyte, 2018-11-12:
@nick-atmosphereiot the captive portal works by a redirecting everything to the esp, so any http request ends.up loading the esp webpage. That has nothing to do with the Android logic for routing.
In any case, I don't see a core issue here, so closing.

closed 2018-11-12 by devyte

rajnish1959, 2019-10-23:
I am no software guy by any means yet tried activating mdns on android by installing printer apps like LetsPrintDroid , Printershare and Printbot which use mdns and lo and behold - both chrome and firefox immediately started recognizing esp8266.local! It may be useful for many of us to simply install these apps so that browsers on android become useful on iot plateforms. May be some of the wise software guys find out which of the three did the trick.