esp8266/Arduino #5316: support to run on flash chips SST26XXXXX

vslinuxdotnet, 2018-11-06:
Hello,
Any support to flash chips SST26VF032B or SST26VF064B from microchip ?

Thank you

vslinuxdotnet, 2018-11-07:
I tested and don't work, I get "A fatal error occurred: MD5 of file does not match data in flash!" from esptool.py but I can read flash ID and format it.

devyte, 2018-11-07:
We don't use esptool.py,  we use esptool-ck, which is altogether different.
Having said that, I'm not aware of the low level details of flash access, but I can say there is no current plan to expand the current support of flash chips.
If you, or someone else, implements it, we can discuss integration.
I'll leave this open for now with Help Wanted.

devyte, 2018-11-07:
One last thing: as a first step, the esptool must support the flash chip, and neitheer esptool ismust hosted in this repo. After the esptool support is implemented, the core may need enhancement in case reading/writing requires different code from what is currently implemented. I suggest figuring out the esptool thing first, and that code is not hosted here.

devyte, 2019-10-31:
@vslinuxdotnet The migration to esptool.py is now done. Will you be looking at this again?

devyte, 2019-11-09:
No feedback in a long while, and in the year this has been open there hasn't been interest by others.
Closing.

devyte, 2019-11-09:
No feedback in a long while, and in the year this has been open there has been no interest by others.
Closing.

closed 2019-11-09 by devyte